FIRST-Tech-Challenge/FtcRobotController #3: Removing gradlew causes travis-ci to fail with wrong version of Gradle

alan412, 2020-09-24:
The removal of gradlew causes travis-ci to fail.   Don't know why it is needed by Travis or why it is no longer in 6.0

FROGbots-4634, 2020-09-24:
Hmm this does look like it was accidentally omitted

cmacfarl, 2020-09-25:
Migration damage.

CalKestis, 2020-09-25:
@alan412 You want to give this branch a spin and see if fixes your travis-ci problem?

https://github.com/FIRST-Tech-Challenge/FtcRobotController/tree/missing-gradlew

alan412, 2020-09-25:
That fixes the missing gradlew problem.   There is still the #4 issue (lint)

I am tempted to propose a simple .travis.yml file to go into the repo - then anyone that has their repo as a public repo in GitHub will get Travis-CI building on all commits "for free".    (My team actually has master protected so you can't push to it and you can't do a pull request until Travis-CI has successfully built it.  A little thing but it makes sure that the master branch always builds.)

Here is the simplest one that will work:

# .travis.yml (put in root dir)
```
language: android
dist: trusty
android:
  components:
    - build-tools-29.0.2
    - android-28
```

PR #7 (2020-09-25): Missing gradle wrappers
Before issuing a pull request, please see the contributing page.

Fixes #3
closed 2020-09-25 by CalKestis