FIRST-Tech-Challenge/FtcRobotController #4: Lint Error

alan412, 2020-09-24:
The lint error that was in 5.5 is still here in 6.0

See: https://github.com/FIRST-Tech-Challenge/SkyStone/issues/262

dansman805, 2020-09-25:
To clarify, this isn't an issue with Travis CI; this is an issue that has to do with the `lint` gradle task. However, this comes up in Travis CI because it likely uses some form of `./gradlew build` under the hood, which would invoke the lint gradle task. This also precludes using `./gradlew build` from the command line in the FTC SDK.

Any call to the `lint` task will cause this error however (in this case caused by running `./gradlew lint`:
```
> Task :FtcRobotController:lint FAILED
Ran lint on variant debug: 34 issues found
Ran lint on variant release: 34 issues found
Wrote HTML report to file:///home/dansman805/robotics/temp-ftc-sdk/FtcRobotController/build/reports/lint-results.html
Wrote XML report to file:///home/dansman805/robotics/temp-ftc-sdk/FtcRobotController/build/reports/lint-results.xml

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':FtcRobotController:lint'.
> Lint found errors in the project; aborting build.

  Fix the issues identified by lint, or add the following to your build script to proceed with errors:
  ...
  android {
      lintOptions {
          abortOnError false
      }
  }
  ...

  The first 3 errors (out of 5) were:
  /home/dansman805/robotics/temp-ftc-sdk/FtcRobotController/src/main/res/menu/ftc_robot_controller.xml:39: Error: Should use app:showAsAction with the appcompat library with xmlns:app="http://schemas.android.com/apk/res-auto" [AppCompatResource]
      android:showAsAction="never"
      ~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  /home/dansman805/robotics/temp-ftc-sdk/FtcRobotController/src/main/res/menu/ftc_robot_controller.xml:44: Error: Should use app:showAsAction with the appcompat library with xmlns:app="http://schemas.android.com/apk/res-auto" [AppCompatResource]
      android:showAsAction="never"
      ~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  /home/dansman805/robotics/temp-ftc-sdk/FtcRobotController/src/main/res/menu/ftc_robot_controller.xml:50: Error: Should use app:showAsAction with the appcompat library with xmlns:app="http://schemas.android.com/apk/res-auto" [AppCompatResource]
      android:showAsAction="never"
      ~~~~~~~~~~~~~~~~~~~~~~~~~~~~

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.
```

Steps for reproduction:
1. `cd` into the sdk's directory from the command line
2. run `./gradlew lint` (or `./gradlew build`!)

Note: this was run on Fedora, however I doubt OS would matter here

Now of course, this isn't really an issue if you don't call `./gradlew lint`, however this also occurs when you run `./gradlew build`, which is the most convienient way to build from the command line and how CI would likely go about doing things.

This can either be fixed by fixing the actual lint errors, or by adding 
```groovy
android {
      lintOptions {
          abortOnError false
      }
  }
```
to the build.gradle of the affected modules.

Of course, actually fixing the lint errors would seem to be the preferred course of action.

shaunsingh, 2020-12-06:
adding 

`android {
      lintOptions {
          abortOnError false
      }
  }`

still results in the same lint errors, 

however, running `./gradlew build -x lint` seemed to work for me

alan412, 2021-09-20:
This error is still there in 7.0

cbroosky, 2021-09-25:
Can confirm, still here in 7.0...

Windwoes, 2021-10-03:
This will be fixed in the next release

Windwoes, 2022-01-17:
Fixed in 7.1

closed 2022-01-17 by Windwoes