FIRST-Tech-Challenge/FtcRobotController #8: The gradlew file isn't executable

dansman805, 2020-09-25:
The `gradlew` bash script doesn't have the executable bit set; while this isn't a big deal, as people who are likely to run `gradlew` probably know enough to make the file executable, it should still be set. 

Running `ls -l gradlew` on my system gets the result of `-rw-r--r--. 1 dansman805 dansman805 5080 Sep 25 13:15 gradlew`.

The lack of an `x` in `-rw-r--r--` indicates that the file is not marked as an executable.

CalKestis, 2020-09-25:
Huh, so I wonder if this is a github/git interaction thing.  Cause on the system in the workspace that push was made from gradlew is 755.

```
-rwxr-xr-x 1 calkestis 1049089  5244 Sep 25 10:01 gradlew
```

dansman805, 2020-09-25:
Looks like adding the file again with `git add --chmod=+x -- gradlew` and then committing it might fix the issue?

https://github.com/git/git/commit/4e55ed32db81d06a4f618e2cc0f9da0e223ae304
 

> **add: add --chmod=+x / --chmod=-x options**
> 
> The executable bit will not be detected (and therefore will not be
> set) for paths in a repository with `core.filemode` set to false,
> though the users may still wish to add files as executable for
> compatibility with other users who _do_ have `core.filemode`
> functionality.  For example, Windows users adding shell scripts may
> wish to add them as executable for compatibility with users on
> non-Windows.
> 
> Although this can be done with a plumbing command
> (`git update-index --add --chmod=+x foo`), teaching the `git-add`
> command allows users to set a file executable with a command that
> they're already familiar with.
>

merscwog, 2020-09-28:
> ```
> -rwxr-xr-x 1 calkestis 1049089  5244 Sep 25 10:01 gradlew
> ```

The wrapper task when run under linux at least, creates a gradlew script that normally has ```-rwxrwxr-x``` permissions, so this already looks suspicious.

It also seems odd that the group id is 1049089 instead of being resolvable to a group name that ```ls``` can display.

Is this a native unix system, or a git-bash shell under windows?

If a unix system, was this copied from some other system first with the same user id but different group ids assigned to the user?

As mentioned, core.filemode might be set to false in your git settings, and regardless, if on a windows system, even when using git-bash, you often have to explicitly do the referenced ```git update-index --add --chmod=+x gradlew``` and push the change up to the main repository.

NoahAndrews, 2021-09-09:
This will be fixed in version 7.0

Windwoes, 2021-09-20:
Fixed in v7.0

closed 2021-09-20 by Windwoes