FIRST-Tech-Challenge/FtcRobotController #12: Consider supporting cross-compilation from newer JDKs without warnings

merscwog, 2020-09-28:
Attempt to compile using JDK 13 or higher complains with:

> warning: [options] source value 7 is obsolete and will be removed in a future release
> warning: [options] target value 7 is obsolete and will be removed in a future release
> warning: [options] To suppress warnings about obsolete options, use -Xlint:-options.

And to make matters even more annoying, it happens for 4-tasks, so you get 12 total warnings every time you build.

Installing an ancient Java 7 JDK is one option, or not using anything newer than a Java 11 JDK for cross-compilation is another, but many systems are switching to installing JDK 13 as the default, and the below code snippet allows newer cross-compilation JDKs without emitting the annoying warnings, yet keeps everything else functionally equivalent:

### build.gradle (top-level one, in the allprojects {} section)
```groovy
    // Avoid warning message from newer JDKs about Java 7 source/target being obsolete
    tasks.withType(JavaCompile).whenTaskAdded { JavaCompile javaCompile ->
        javaCompile.options.compilerArgs << '-Xlint:-options'
    }
```

The snippet directly replaced in the build.gradle file can be seen here:
https://github.com/FIRST-Tech-Challenge/FtcRobotController/pull/11/files

FROGbots-4634, 2020-09-28:
> Installing an ancient Java 7 JDK is one option, or not using anything newer than a Java 11 JDK for cross-compilation is another, but many systems are switching to installing JDK 13 as the default,

I'm confused. IntelliJ and Android Studio now ship with a bundled [JetBrainsRuntime](https://confluence.jetbrains.com/display/JBR/JetBrains+Runtime) version of the JDK, which fixes known bugs in the Oracle/OpenJDK. Does this issue still happen when using the JetBrainsRuntime?

merscwog, 2020-09-28:
I'm using IntelliJ Community Edition 2020.2.2 under Ubuntu, and indeed, there is the normal JetBrain's Runtime which is openjdk 11 based, but is not a selectable option for Java SDK usage for projects.

This makes sense, because it's intended to be just a JRE and the current [IntelliJ FAQ SDK section](https://www.jetbrains.com/help/idea/sdk.html#jdk) has this to say:

> **Important notice**
> The bundled JRE is used for running the IDE itself, and it's not sufficient for developing Java applications. Before you start developing in Java, download and install a standalone JDK build.
> 

Even trying to manually add a JDK that points to the directory containing their JRE (which does contain a javac command) refuses to accept it as a valid Java home.

FROGbots-4634, 2020-09-28:
Hmm, OK. But I don't think Android Studio requires a standalone JDK as I've been able to run it on systems without a standalone JDK, anyway :)

alan412, 2020-09-28:
Android Studio does not require a standalone JDK.  However, we install one
for our students so they can run Beta8397's virtual robot simulator.

--Alan