FIRST-Tech-Challenge/FtcRobotController #13: MR Core Control Module support

AlecHub, 2020-09-29:
Our outreach bots are using Modern Robotics Core Control Modules.  Are MR modules still supported in the latest RobotController SDK?

FROGbots-4634, 2020-09-29:
I think they are, for now, but given that MR is illegal for competition, I wouldn't be surprised if support was broken in future updates this season.

AlecHub, 2020-09-29:
Ok thanks.  We could always use an older version of the SDK if and when support for MR modules is removed.  It would be more convenient if support were not removed.

JIceberg, 2020-09-29:
Looks like there is still support if you look at the [Extracted-RC](https://github.com/OpenFTC/Extracted-RC/tree/master/RobotCore/src/main/java/com/qualcomm/robotcore/hardware), courtesy of OpenFTC, which is linked in the [bintray](https://bintray.com/first-tech-challenge/ftcsdk/RobotCore) for the SDK modules. Since the resources are now on bintray, you can always use the most recent version of support for the modules (i.e. 6.0.1 as of now).

ftctechnh, 2020-10-01:
@AlecHub Unfortunately you will have to use an older version of the SDK if you'd like to use the MR modules for your outreach or demo bots.

closed 2020-10-01 by ftctechnh

ftctechnh, 2020-10-01:
@AlecHub I was incorrect.  Apparently the support for the older MR core control modules is still included in v5.5 and v6.0 versions of the software.  You should be able to use the current software with the older MR modules.