FIRST-Tech-Challenge/FtcRobotController #14: Dualshock 4 connection issues?

tjhawkey, 2020-09-30:
Hmm. No luck making one work. Can you post instructions? I found the key mapping, but the 'Start  A", equivalent doesn't work and there's no obvious way to make the controller connect. Setup works fine with F310 and gamepad and cable have been tested.
Thanks!

FROGbots-4634, 2020-09-30:
If you go to the Gamepad Type Overrides menu on the driver station and specifically tell it that the gamepad is an PS4 gamepad, does it work?

Also, what phone and Android version?

tjhawkey, 2020-09-30:
Tested so far with moto E4, Android 7.1.1

FROGbots-4634, 2020-09-30:
Interesting that you say it doesn't pick up the key press. Do any of the buttons (stick or dpad) work in other parts of the Android OS? For instance can you select an all in the app tray using the gamepad?

tjhawkey, 2020-09-30:
Good question. The phone doesn't see any key presses at all (moving around app tray). F310 works fine. I'll find a newer phone to test with.

ftctechnh, 2020-09-30:
@tjhawkey I just retested this.  I am using a Sony PS4 gamepad, that is connected through a USB Type A to Type B micro cable, which is plugged into on OTG Micro USB adapter.  I have an E4, Android 7.1.1 and plugged it in.  If I designate driver 1 i see the User 1 icon appear and if i check the button presses and joystick moves on the RC, everything seems OK.

same thing happens if i designate the gamepad as driver 2.

have you check to see if the cable setup is good? Maybe the adapter or the USB cable is worn? Also, i'm not sure if this is required, but is your PS4 gamepad charged up?

Here's a photo of my DS setup.

[image]

tjhawkey, 2020-09-30:
Hello. Thanks for checking! I am pretty sure that the gamepad is charged, but I'll double check that. I'll also check the cables again.
Can you confirm the designation key sequence?
Seems more like a connection issue (somehow), or maybe I got a counterfeit controller!
Odd that it works connected to my PC fine.

FROGbots-4634, 2020-09-30:
Can you provide the [USB VID and PID](https://superuser.com/questions/1106247/how-can-i-get-the-vendor-id-and-product-id-for-a-usb-device) that the controller reports when plugged into your computer?

tjhawkey, 2020-09-30:
Works fine with a different controller. Not sure what's up with that, but at least it works. The selection button for which station is the one on the right (different than the F310)
Thanks!

ftctechnh, 2020-10-01:
@tjhawkey When you say "Works fine with a _different controller_" does that mean with a different PS4 controller?

Also, the selection button on the F310 controller is the "Start" button which is analogous to the "Options" button on the PS4.  Both buttons are on the center right portion of their respective game controllers.

tjhawkey, 2020-10-01:
Thanks for the reply. I meant as in "with a known official Sony PS4 controller". Suspect the non-connecting one is counterfeit. I guess that's a thing.

ftctechnh, 2020-10-01:
@tjhawkey Thanks for the clarification!

closed 2020-10-01 by ftctechnh

tjhawkey, 2020-10-01:
Any outlook for using the haptic part of the controllers at some point? That would be pretty spiffy.

JAKCODING, 2020-10-09:
@FROGbots-4634 how would we write a HID driver that works within the FTC framework? As far as I know, the only way to do that would be to modify the Driver Station app, but I could be completely wrong...

FROGbots-4634, 2020-10-09:
> As far as I know, the only way to do that would be to modify the Driver Station app, but I could be completely wrong...

That's correct.

JAKCODING, 2020-10-09:
Just for fun... how would I do that?