FIRST-Tech-Challenge/FtcRobotController #15: OpMode Selectable via Robot Controller phone

AlecHub, 2020-09-30:
We would like to be able to select, run, and stop a [special class of] OpMode via the Robot Controller phone.  The propose is to prototype and demo mechanisms without the need of a Driver Station.  

The OpMode would be controlled via digital and analog ports (i.e. switches and pots) on the controller network.  Telemetry would be displayed on the Robot Controller phone.  

Support for gamepads connected to Robot Controller phone would be nice, but not necessary.

FROGbots-4634, 2020-10-02:
FWIW, safety concerns aside, using OpenRC Turbo you could modify the default OpMode and change it from StopRobot to something else.

AlecHub, 2020-10-03:
> FWIW, safety concerns aside, using OpenRC Turbo you could modify the default OpMode and change it from StopRobot to something else.

That's worth a lot, thanks.

With the stock SDK, if I have a StopRobot OpMode class in my TeamCode, does that override the SDK's StopRobot?

FROGbots-4634, 2020-10-03:
No, that would not work. There are a couple places which directly reference the OpModeManagerImpl.DefaultOpMode class, so you will need to use OpenRC to be able to modify those.

cmacfarl, 2021-09-25:
A combined DS/RC has periodically been discussed over the years, beginning with the first year using the Android platform.  For a number of reasons the decision has always been not to implement this feature.

closed 2021-09-25 by cmacfarl