FIRST-Tech-Challenge/FtcRobotController #19: Unexpected exception thrown during lynx communication

odpgit, 2020-10-04:
We are occasionally seeing this error during testing with the new SDK and Roadrunner. It seems to be causing accuracy issues with odometry-based positioning.

Here is the build.release.gradle for TeamCode:

dependencies {
    implementation project(':FtcRobotController')
    implementation 'org.firstinspires.ftc:RobotCore:6.0.1'
    implementation 'org.firstinspires.ftc:Hardware:6.0.1'
    implementation 'org.firstinspires.ftc:FtcCommon:6.0.1'
    implementation (name: 'tfod-release', ext:'aar')
    implementation (name: 'tensorflow-lite-0.0.0-nightly', ext:'aar')

    implementation 'org.apache.commons:commons-math3:3.6.1'

    implementation 'com.acmerobotics.roadrunner:core:0.5.2'
    implementation 'com.acmerobotics.dashboard:dashboard:0.3.10'
}

And here is the error:

2020-09-27 17:29:13.791 5236-5394/com.qualcomm.ftcrobotcontroller E/LynxModule: unexpected exception thrown during lynx communication
2020-09-27 17:29:13.792 5236-5394/com.qualcomm.ftcrobotcontroller E/LynxModule: java.lang.InterruptedException
2020-09-27 17:29:13.793 5236-5363/com.qualcomm.ftcrobotcontroller V/ThreadPool: waiting for service user linear op mode
2020-09-27 17:29:13.795 5236-5394/com.qualcomm.ftcrobotcontroller E/LynxModule:     at java.util.concurrent.locks.AbstractQueuedSynchronizer.doAcquireSharedNanos(AbstractQueuedSynchronizer.java:1036)
2020-09-27 17:29:13.796 5236-5394/com.qualcomm.ftcrobotcontroller E/LynxModule:     at java.util.concurrent.locks.AbstractQueuedSynchronizer.tryAcquireSharedNanos(AbstractQueuedSynchronizer.java:1327)
2020-09-27 17:29:13.797 5236-5394/com.qualcomm.ftcrobotcontroller E/LynxModule:     at java.util.concurrent.CountDownLatch.await(CountDownLatch.java:249)
2020-09-27 17:29:13.798 5236-5394/com.qualcomm.ftcrobotcontroller E/LynxModule:     at com.qualcomm.hardware.lynx.commands.LynxRespondable.awaitAndRetransmit(LynxRespondable.java:385)
2020-09-27 17:29:13.798 5236-5394/com.qualcomm.ftcrobotcontroller E/LynxModule:     at com.qualcomm.hardware.lynx.commands.LynxRespondable.awaitAckResponseOrNack(LynxRespondable.java:400)
2020-09-27 17:29:13.799 5236-5394/com.qualcomm.ftcrobotcontroller E/LynxModule:     at com.qualcomm.hardware.lynx.commands.LynxRespondable.sendReceive(LynxRespondable.java:259)
2020-09-27 17:29:13.800 5236-5394/com.qualcomm.ftcrobotcontroller E/LynxModule:     at com.qualcomm.hardware.lynx.LynxModule.getBulkData(LynxModule.java:1395)
2020-09-27 17:29:13.801 5236-5394/com.qualcomm.ftcrobotcontroller E/LynxModule:     at com.qualcomm.hardware.lynx.LynxModule.recordBulkCachingCommandIntent(LynxModule.java:1508)
2020-09-27 17:29:13.802 5236-5394/com.qualcomm.ftcrobotcontroller E/LynxModule:     at com.qualcomm.hardware.lynx.LynxModule.recordBulkCachingCommandIntent(LynxModule.java:1478)
2020-09-27 17:29:13.803 5236-5394/com.qualcomm.ftcrobotcontroller E/LynxModule:     at com.qualcomm.hardware.lynx.LynxDcMotorController.getMotorCurrentPosition(LynxDcMotorController.java:719)
2020-09-27 17:29:13.804 5236-5394/com.qualcomm.ftcrobotcontroller E/LynxModule:     at com.qualcomm.robotcore.hardware.DcMotorImpl.getCurrentPosition(DcMotorImpl.java:293)
2020-09-27 17:29:13.805 5236-5394/com.qualcomm.ftcrobotcontroller E/LynxModule:     at org.firstinspires.ftc.teamcode.util.Encoder.getCurrentPosition(Encoder.java:74)
2020-09-27 17:29:13.806 5236-5394/com.qualcomm.ftcrobotcontroller E/LynxModule:     at org.firstinspires.ftc.teamcode.drive.StandardTrackingWheelLocalizer.getWheelPositions(StandardTrackingWheelLocalizer.java:65)
2020-09-27 17:29:13.807 5236-5394/com.qualcomm.ftcrobotcontroller E/LynxModule:     at com.acmerobotics.roadrunner.localization.ThreeTrackingWheelLocalizer.update(ThreeTrackingWheelLocalizer.kt:60)
2020-09-27 17:29:13.807 5236-5394/com.qualcomm.ftcrobotcontroller E/LynxModule:     at com.acmerobotics.roadrunner.drive.Drive.updatePoseEstimate(Drive.kt:53)
2020-09-27 17:29:13.808 5236-5394/com.qualcomm.ftcrobotcontroller E/LynxModule:     at org.firstinspires.ftc.teamcode.drive.SampleMecanumDrive.update(SampleMecanumDrive.java:212)
2020-09-27 17:29:13.809 5236-5394/com.qualcomm.ftcrobotcontroller E/LynxModule:     at org.firstinspires.ftc.teamcode.drive.opmode.LocalizationTest.runOpMode(LocalizationTest.java:71)
2020-09-27 17:29:13.810 5236-5394/com.qualcomm.ftcrobotcontroller E/LynxModule:     at com.qualcomm.robotcore.eventloop.opmode.LinearOpMode$LinearOpModeHelper$1.run(LinearOpMode.java:271)
2020-09-27 17:29:13.811 5236-5394/com.qualcomm.ftcrobotcontroller E/LynxModule:     at com.qualcomm.robotcore.util.ThreadPool.logThreadLifeCycle(ThreadPool.java:737)
2020-09-27 17:29:13.812 5236-5394/com.qualcomm.ftcrobotcontroller E/LynxModule:     at com.qualcomm.robotcore.eventloop.opmode.LinearOpMode$LinearOpModeHelper.run(LinearOpMode.java:266)
2020-09-27 17:29:13.813 5236-5394/com.qualcomm.ftcrobotcontroller E/LynxModule:     at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1133)
2020-09-27 17:29:13.813 5236-5394/com.qualcomm.ftcrobotcontroller E/LynxModule:     at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607)
2020-09-27 17:29:13.814 5236-5394/com.qualcomm.ftcrobotcontroller E/LynxModule:     at com.qualcomm.robotcore.util.ThreadPool$ThreadFactoryImpl$1.run(ThreadPool.java:793)
2020-09-27 17:29:13.815 5236-5394/com.qualcomm.ftcrobotcontroller E/LynxModule:     at java.lang.Thread.run(Thread.java:761)

FROGbots-4634, 2020-10-04:
This is totally normal, notice that it's an interrupted exception:

```
 E/LynxModule: java.lang.InterruptedException
```

That will happen when you press stop or the timer expires while the OpMode is running.

NoahAndrews, 2021-01-30:
The next version of the Robot Controller will not have this logspam.

Windwoes, 2021-02-23:
SDK v6.2 has been released, and no longer spams the log with interrupted exception stack traces when stop is requested from the Driver Station.

closed 2021-02-23 by Windwoes