FIRST-Tech-Challenge/FtcRobotController #21: Unable to use USB Webcam with Control Hub

smartynenko, 2020-10-07:
I am getting exception when I attempt to run ConceptWebcam example with external USB camera connected to Control Hub (exception details are below). Vuforia appear be be able to recognize camera and stream video with no issues. Please advise

2020-10-07 14:55:29.303 2338-2487/com.qualcomm.ftcrobotcontroller D/UsbResiliantWebcam|DelCaptureSession: camera session is configured: Webcam(1118:1833:3-1.4)
2020-10-07 14:55:29.304 2338-2489/com.qualcomm.ftcrobotcontroller V/UvcApiCameraCaptureSequence: construct(0x08e4bbb8)
2020-10-07 14:55:29.304 2338-2489/com.qualcomm.ftcrobotcontroller D/UvcDeviceHandle: getStreamControl(320x240 15)...
2020-10-07 14:55:29.305 2338-2489/com.qualcomm.ftcrobotcontroller V/UvcStreamCtrl: construct(0x02e62d91)
2020-10-07 14:55:29.305 2338-2489/com.qualcomm.ftcrobotcontroller D/Uvc: [jni_devicehandle.cpp:115] Java_org_firstinspires_ftc_robotcore_internal_camera_libuvc_nativeobject_UvcDeviceHandle_nativeGetStreamControlFormatSize()...
2020-10-07 14:55:29.305 2338-2489/com.qualcomm.ftcrobotcontroller V/Uvc: API: Java_org_firstinspires_ftc_robotcore_internal_camera_libuvc_nativeobject_UvcDeviceHandle_nativeGetStreamControlFormatSize: enter...
2020-10-07 14:55:29.305 2338-2489/com.qualcomm.ftcrobotcontroller D/Uvc: [stream.cpp:373] uvc_get_stream_ctrl_format_size()...
2020-10-07 14:55:29.305 2338-2489/com.qualcomm.ftcrobotcontroller D/Uvc: [libuvc/libuvc_internal.h:770] uvc_device_handle::claimInterface(idx=1)...
2020-10-07 14:55:29.305 2338-2489/com.qualcomm.ftcrobotcontroller D/UvcDevice: claimInterface(1)
2020-10-07 14:55:29.305 2338-2489/com.qualcomm.ftcrobotcontroller D/UvcDevice: claimInterface(1) succeeded
2020-10-07 14:55:29.305 2338-2489/com.qualcomm.ftcrobotcontroller D/Uvc: [libuvc/libuvc_internal.h:783] claimInterface(1) succeeded
2020-10-07 14:55:29.305 2338-2489/com.qualcomm.ftcrobotcontroller D/Uvc: [libuvc/libuvc_internal.h:770] ...uvc_device_handle::claimInterface() rc=0
2020-10-07 14:55:29.305 2338-2489/com.qualcomm.ftcrobotcontroller D/Uvc: [libuvc/libuvc_internal.h:793] uvc_device_handle::releaseInterface()...
2020-10-07 14:55:29.306 2338-2489/com.qualcomm.ftcrobotcontroller D/UvcDevice: releaseInterface(1)
2020-10-07 14:55:29.306 2338-2489/com.qualcomm.ftcrobotcontroller D/UvcDevice: releaseInterface(1) succeeded
2020-10-07 14:55:29.306 2338-2489/com.qualcomm.ftcrobotcontroller D/Uvc: [libuvc/libuvc_internal.h:793] ...uvc_device_handle::releaseInterface() rc=0
2020-10-07 14:55:29.306 2338-2489/com.qualcomm.ftcrobotcontroller D/Uvc: [libuvc.h:147] [stream.cpp:470] originating err: -51(UVC_ERROR_INVALID_MODE)
2020-10-07 14:55:29.306 2338-2489/com.qualcomm.ftcrobotcontroller D/Uvc: [stream.cpp:373] ...uvc_get_stream_ctrl_format_size() rc=-51
2020-10-07 14:55:29.306 2338-2489/com.qualcomm.ftcrobotcontroller V/Uvc: API: Java_org_firstinspires_ftc_robotcore_internal_camera_libuvc_nativeobject_UvcDeviceHandle_nativeGetStreamControlFormatSize: ...exit
2020-10-07 14:55:29.306 2338-2489/com.qualcomm.ftcrobotcontroller D/Uvc: [jni_devicehandle.cpp:115] ...Java_org_firstinspires_ftc_robotcore_internal_camera_libuvc_nativeobject_UvcDeviceHandle_nativeGetStreamControlFormatSize()
2020-10-07 14:55:29.306 2338-2489/com.qualcomm.ftcrobotcontroller V/UvcStreamCtrl: destroy(pointer=0x70389469c0)
2020-10-07 14:55:29.307 2338-2489/com.qualcomm.ftcrobotcontroller D/UvcDeviceHandle: ...getStreamControl(320x240 15)
2020-10-07 14:55:29.307 2338-2489/com.qualcomm.ftcrobotcontroller E/UvcDeviceHandle: getStreamControl() failed
2020-10-07 14:55:29.307 2338-2489/com.qualcomm.ftcrobotcontroller D/UvcApiCaptureSession: stopCapture()...
2020-10-07 14:55:29.308 2338-2489/com.qualcomm.ftcrobotcontroller V/UvcApiCameraCaptureSequence: destroy(hash=0x08e4bbb8)
2020-10-07 14:55:29.308 2338-2489/com.qualcomm.ftcrobotcontroller D/UvcApiCaptureSession: ...stopCapture()
2020-10-07 14:55:29.308 2338-2489/com.qualcomm.ftcrobotcontroller E/UsbResiliantWebcam|DelCaptureSequence: exception starting capture
2020-10-07 14:55:29.311 2338-2489/com.qualcomm.ftcrobotcontroller E/UsbResiliantWebcam|DelCaptureSequence: org.firstinspires.ftc.robotcore.external.hardware.camera.CameraException
2020-10-07 14:55:29.316 2338-2489/com.qualcomm.ftcrobotcontroller E/UsbResiliantWebcam|DelCaptureSequence:     at org.firstinspires.ftc.robotcore.internal.camera.libuvc.api.UvcApiCameraCaptureSequence.startStreaming(UvcApiCameraCaptureSequence.java:177)
2020-10-07 14:55:29.319 2338-2489/com.qualcomm.ftcrobotcontroller E/UsbResiliantWebcam|DelCaptureSequence:     at org.firstinspires.ftc.robotcore.internal.camera.libuvc.api.UvcApiCaptureSession.startCapture(UvcApiCaptureSession.java:203)
2020-10-07 14:55:29.322 2338-2489/com.qualcomm.ftcrobotcontroller E/UsbResiliantWebcam|DelCaptureSequence:     at org.firstinspires.ftc.robotcore.internal.camera.delegating.DelegatingCaptureSequence$5.run(DelegatingCaptureSequence.java:330)
2020-10-07 14:55:29.324 2338-2489/com.qualcomm.ftcrobotcontroller E/UsbResiliantWebcam|DelCaptureSequence:     at org.firstinspires.ftc.robotcore.internal.system.Tracer.trace(Tracer.java:243)
2020-10-07 14:55:29.327 2338-2489/com.qualcomm.ftcrobotcontroller E/UsbResiliantWebcam|DelCaptureSequence:     at org.firstinspires.ftc.robotcore.internal.camera.delegating.DelegatingCaptureSequence.doStreaming(DelegatingCaptureSequence.java:314)
2020-10-07 14:55:29.330 2338-2489/com.qualcomm.ftcrobotcontroller E/UsbResiliantWebcam|DelCaptureSequence:     at org.firstinspires.ftc.robotcore.internal.camera.delegating.DelegatingCaptureSequence$2.run(DelegatingCaptureSequence.java:163)
2020-10-07 14:55:29.331 2338-2489/com.qualcomm.ftcrobotcontroller E/UsbResiliantWebcam|DelCaptureSequence:     at org.firstinspires.ftc.robotcore.internal.system.Tracer.trace(Tracer.java:243)
2020-10-07 14:55:29.333 2338-2489/com.qualcomm.ftcrobotcontroller E/UsbResiliantWebcam|DelCaptureSequence:     at org.firstinspires.ftc.robotcore.internal.camera.delegating.DelegatingCaptureSequence.startStreaming(DelegatingCaptureSequence.java:147)
2020-10-07 14:55:29.334 2338-2489/com.qualcomm.ftcrobotcontroller E/UsbResiliantWebcam|DelCaptureSequence:     at org.firstinspires.ftc.robotcore.internal.camera.delegating.DelegatingCaptureSession$6.get(DelegatingCaptureSession.java:273)
2020-10-07 14:55:29.335 2338-2489/com.qualcomm.ftcrobotcontroller E/UsbResiliantWebcam|DelCaptureSequence:     at org.firstinspires.ftc.robotcore.internal.camera.delegating.DelegatingCaptureSession$6.get(DelegatingCaptureSession.java:256)
2020-10-07 14:55:29.336 2338-2489/com.qualcomm.ftcrobotcontroller E/UsbResiliantWebcam|DelCaptureSequence:     at org.firstinspires.ftc.robotcore.internal.system.Tracer.trace(Tracer.java:230)
2020-10-07 14:55:29.337 2338-2489/com.qualcomm.ftcrobotcontroller E/UsbResiliantWebcam|DelCaptureSequence:     at org.firstinspires.ftc.robotcore.internal.camera.delegating.DelegatingCaptureSession.startCapture(DelegatingCaptureSession.java:255)
2020-10-07 14:55:29.338 2338-2489/com.qualcomm.ftcrobotcontroller E/UsbResiliantWebcam|DelCaptureSequence:     at org.firstinspires.ftc.robotcore.internal.camera.delegating.DelegatingCaptureSession.startCapture(DelegatingCaptureSession.java:248)
2020-10-07 14:55:29.339 2338-2489/com.qualcomm.ftcrobotcontroller E/UsbResiliantWebcam|DelCaptureSequence:     at org.openftc.easyopencv.OpenCvWebcamImpl$3.onConfigured(OpenCvWebcamImpl.java:323)
2020-10-07 14:55:29.340 2338-2489/com.qualcomm.ftcrobotcontroller E/UsbResiliantWebcam|DelCaptureSequence:     at org.firstinspires.ftc.robotcore.internal.camera.delegating.DelegatingCaptureSession$4.handle(DelegatingCaptureSession.java:208)
2020-10-07 14:55:29.341 2338-2489/com.qualcomm.ftcrobotcontroller E/UsbResiliantWebcam|DelCaptureSequence:     at org.firstinspires.ftc.robotcore.internal.camera.delegating.DelegatingCaptureSession$4.handle(DelegatingCaptureSession.java:205)
2020-10-07 14:55:29.341 2338-2489/com.qualcomm.ftcrobotcontroller E/UsbResiliantWebcam|DelCaptureSequence:     at org.firstinspires.ftc.robotcore.external.function.Continuation$ExecutorDispatcher$1.run(Continuation.java:151)
2020-10-07 14:55:29.342 2338-2489/com.qualcomm.ftcrobotcontroller E/UsbResiliantWebcam|DelCaptureSequence:     at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1133)
2020-10-07 14:55:29.343 2338-2489/com.qualcomm.ftcrobotcontroller E/UsbResiliantWebcam|DelCaptureSequence:     at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607)
[... cut ...]
2020-10-07 14:55:29.346 2338-2489/com.qualcomm.ftcrobotcontroller D/UsbResiliantWebcam|DelCaptureSequence: undoStreaming()...
2020-10-07 14:55:29.346 2338-2489/com.qualcomm.ftcrobotcontroller D/UvcApiCaptureSession: doClose()...
2020-10-07 14:55:29.346 2338-2489/com.qualcomm.ftcrobotcontroller D/UvcApiCaptureSession: ...doClose()
2020-10-07 14:55:29.347 2338-2489/com.qualcomm.ftcrobotcontroller D/UsbResiliantWebcam|DelCaptureSequence: ...undoStreaming()
2020-10-07 14:55:29.347 2338-2487/com.qualcomm.ftcrobotcontroller D/UsbResiliantWebcam|DelCaptureSession: camera session is closed: Webcam(1118:1833:3-1.4)

smartynenko, 2020-10-07:
rebuilt app from scratch, removing all references to easycv, still fails with slightly different error:

2020-10-07 16:18:47.820 1425-1448/com.qualcomm.ftcrobotcontroller E/UvcDeviceHandle: getStreamControl() failed
2020-10-07 16:18:47.821 1425-1448/com.qualcomm.ftcrobotcontroller D/UvcApiCaptureSession: stopCapture()...
2020-10-07 16:18:47.821 1425-1448/com.qualcomm.ftcrobotcontroller V/UvcApiCameraCaptureSequence: destroy(hash=0x0c966414)
2020-10-07 16:18:47.821 1425-1448/com.qualcomm.ftcrobotcontroller D/UvcApiCaptureSession: ...stopCapture()
2020-10-07 16:18:47.821 1425-1448/com.qualcomm.ftcrobotcontroller E/UsbResiliantWebcam|DelCaptureSequence: exception starting capture
2020-10-07 16:18:47.825 1425-1448/com.qualcomm.ftcrobotcontroller E/UsbResiliantWebcam|DelCaptureSequence: org.firstinspires.ftc.robotcore.external.hardware.camera.CameraException
2020-10-07 16:18:47.831 1425-1448/com.qualcomm.ftcrobotcontroller E/UsbResiliantWebcam|DelCaptureSequence:     at org.firstinspires.ftc.robotcore.internal.camera.libuvc.api.UvcApiCameraCaptureSequence.startStreaming(UvcApiCameraCaptureSequence.java:177)
2020-10-07 16:18:47.834 1425-1448/com.qualcomm.ftcrobotcontroller E/UsbResiliantWebcam|DelCaptureSequence:     at org.firstinspires.ftc.robotcore.internal.camera.libuvc.api.UvcApiCaptureSession.startCapture(UvcApiCaptureSession.java:203)
2020-10-07 16:18:47.838 1425-1448/com.qualcomm.ftcrobotcontroller E/UsbResiliantWebcam|DelCaptureSequence:     at org.firstinspires.ftc.robotcore.internal.camera.delegating.DelegatingCaptureSequence$5.run(DelegatingCaptureSequence.java:330)
2020-10-07 16:18:47.841 1425-1448/com.qualcomm.ftcrobotcontroller E/UsbResiliantWebcam|DelCaptureSequence:     at org.firstinspires.ftc.robotcore.internal.system.Tracer.trace(Tracer.java:243)
2020-10-07 16:18:47.844 1425-1448/com.qualcomm.ftcrobotcontroller E/UsbResiliantWebcam|DelCaptureSequence:     at org.firstinspires.ftc.robotcore.internal.camera.delegating.DelegatingCaptureSequence.doStreaming(DelegatingCaptureSequence.java:314)
2020-10-07 16:18:47.848 1425-1448/com.qualcomm.ftcrobotcontroller E/UsbResiliantWebcam|DelCaptureSequence:     at org.firstinspires.ftc.robotcore.internal.camera.delegating.DelegatingCaptureSequence$2.run(DelegatingCaptureSequence.java:163)
2020-10-07 16:18:47.850 1425-1448/com.qualcomm.ftcrobotcontroller E/UsbResiliantWebcam|DelCaptureSequence:     at org.firstinspires.ftc.robotcore.internal.system.Tracer.trace(Tracer.java:243)
2020-10-07 16:18:47.852 1425-1448/com.qualcomm.ftcrobotcontroller E/UsbResiliantWebcam|DelCaptureSequence:     at org.firstinspires.ftc.robotcore.internal.camera.delegating.DelegatingCaptureSequence.startStreaming(DelegatingCaptureSequence.java:147)
2020-10-07 16:18:47.854 1425-1448/com.qualcomm.ftcrobotcontroller E/UsbResiliantWebcam|DelCaptureSequence:     at org.firstinspires.ftc.robotcore.internal.camera.delegating.DelegatingCaptureSession$6.get(DelegatingCaptureSession.java:273)
2020-10-07 16:18:47.855 1425-1448/com.qualcomm.ftcrobotcontroller E/UsbResiliantWebcam|DelCaptureSequence:     at org.firstinspires.ftc.robotcore.internal.camera.delegating.DelegatingCaptureSession$6.get(DelegatingCaptureSession.java:256)
2020-10-07 16:18:47.856 1425-1448/com.qualcomm.ftcrobotcontroller E/UsbResiliantWebcam|DelCaptureSequence:     at org.firstinspires.ftc.robotcore.internal.system.Tracer.trace(Tracer.java:230)
2020-10-07 16:18:47.857 1425-1448/com.qualcomm.ftcrobotcontroller E/UsbResiliantWebcam|DelCaptureSequence:     at org.firstinspires.ftc.robotcore.internal.camera.delegating.DelegatingCaptureSession.startCapture(DelegatingCaptureSession.java:255)
2020-10-07 16:18:47.858 1425-1448/com.qualcomm.ftcrobotcontroller E/UsbResiliantWebcam|DelCaptureSequence:     at org.firstinspires.ftc.robotcore.internal.camera.delegating.DelegatingCaptureSession.startCapture(DelegatingCaptureSession.java:248)
2020-10-07 16:18:47.859 1425-1448/com.qualcomm.ftcrobotcontroller E/UsbResiliantWebcam|DelCaptureSequence:     at org.firstinspires.ftc.teamcode.ConceptWebcam$2.onConfigured(ConceptWebcam.java:219)
2020-10-07 16:18:47.859 1425-1448/com.qualcomm.ftcrobotcontroller E/UsbResiliantWebcam|DelCaptureSequence:     at org.firstinspires.ftc.robotcore.internal.camera.delegating.DelegatingCaptureSession$4.handle(DelegatingCaptureSession.java:208)
2020-10-07 16:18:47.860 1425-1448/com.qualcomm.ftcrobotcontroller E/UsbResiliantWebcam|DelCaptureSequence:     at org.firstinspires.ftc.robotcore.internal.camera.delegating.DelegatingCaptureSession$4.handle(DelegatingCaptureSession.java:205)
2020-10-07 16:18:47.861 1425-1448/com.qualcomm.ftcrobotcontroller E/UsbResiliantWebcam|DelCaptureSequence:     at org.firstinspires.ftc.robotcore.external.function.Continuation$HandlerDispatcher$1.run(Continuation.java:121)
2020-10-07 16:18:47.862 1425-1448/com.qualcomm.ftcrobotcontroller E/UsbResiliantWebcam|DelCaptureSequence:     at android.os.Handler.handleCallback(Handler.java:755)
2020-10-07 16:18:47.862 1425-1448/com.qualcomm.ftcrobotcontroller E/UsbResiliantWebcam|DelCaptureSequence:     at android.os.Handler.dispatchMessage(Handler.java:95)
2020-10-07 16:18:47.863 1425-1448/com.qualcomm.ftcrobotcontroller E/UsbResiliantWebcam|DelCaptureSequence:     at android.os.Looper.loop(Looper.java:154)
2020-10-07 16:18:47.864 1425-1448/com.qualcomm.ftcrobotcontroller E/UsbResiliantWebcam|DelCaptureSequence:     at org.firstinspires.ftc.robotcore.internal.network.CallbackLooper$1.run(CallbackLooper.java:139)
[... cut ...]
2020-10-07 16:18:47.866 1425-1448/com.qualcomm.ftcrobotcontroller E/UsbResiliantWebcam|DelCaptureSequence:     at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607)
2020-10-07 16:18:47.867 1425-1448/com.qualcomm.ftcrobotcontroller E/UsbResiliantWebcam|DelCaptureSequence:     at com.qualcomm.robotcore.util.ThreadPool$ThreadFactoryImpl$1.run(ThreadPool.java:793)
2020-10-07 16:18:47.867 1425-1448/com.qualcomm.ftcrobotcontroller E/UsbResiliantWebcam|DelCaptureSequence:     at java.lang.Thread.run(Thread.java:761)
2020-10-07 16:18:47.868 1425-1448/com.qualcomm.ftcrobotcontroller D/UsbResiliantWebcam|DelCaptureSequence: undoStreaming()...

FROGbots-4634, 2020-10-07:
Which webcam are you using?

FROGbots-4634, 2020-10-07:
A quick glance at the logs seems to indicate that the camera you're using doesn't want to stream 320x240@15FPS in YUY2.

```
D/Uvc: [libuvc.h:147] [stream.cpp:470] originating err: -51(UVC_ERROR_INVALID_MODE)
D/Uvc: [stream.cpp:373] ...uvc_get_stream_ctrl_format_size() rc=-51
.........
D/UvcDeviceHandle: ...getStreamControl(320x240 15)
E/UvcDeviceHandle: getStreamControl() failed
```

When you say `removing all references to easycv`, does this mean you also tried using EasyOpenCV? If so, did that also fail?

smartynenko, 2020-10-07:
> Which webcam are you using?

Microsoft LifeCam Show

smartynenko, 2020-10-07:
> When you say `removing all references to easycv`, does this mean you also tried using EasyOpenCV? If so, did that also fail?

Correct, tried to use it with EasyOpenCV, but could not make past initialization

FROGbots-4634, 2020-10-07:
Quite interesting that Vuforia worked, as all 3 (Vuforia, ConceptWebcam, and EasyOpenCV) use the same core UVC driver. Can you try something? Can you try EasyOpenCV again, but try resolutions other than 320x240? For instance, 640x480. If you type in some random (unsupported) resolution, EasyOpenCV will crash your program, but will provide a list of all resolutions the webcam claims to support. If you could try each of those that would be great.

smartynenko, 2020-10-07:
Thank you for support, vuforia put corresponding entry into the log file, I was able to start stream with these parameters

10-07 17:13:36.830  3979  4005 D UvcDeviceHandle: getStreamControl(800x600 14)

FROGbots-4634, 2020-10-07:
Very strange. If you have access to a Linux box I'd be curious if `guvcview` can stream from your webcam at 320x240 in YUY2.

smartynenko, 2020-10-07:
I do not have a linux box available at the moment :/

smartynenko, 2020-10-08:
Please refer to this thread for additional details [here](https://github.com/OpenFTC/EasyOpenCV/pull/19#issuecomment-705607444)

Windwoes, 2020-12-26:
I think this can be closed, the issue was fixed in SDK v6.1

closed 2020-12-27 by cmacfarl