FIRST-Tech-Challenge/FtcRobotController #24: Robot Controller App Crashed on startup

technova12611, 2020-10-08:
We downloaded the robot controller app and installed it. It kept crashing when we open the app. The stack trace is attached below.

Phone: Moto E4
Android: 7.1.1

```
2020-10-08 11:59:33.299 9480-9480/com.qualcomm.ftcrobotcontroller E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.qualcomm.ftcrobotcontroller, PID: 9480
    java.lang.RuntimeException: Unable to start activity ComponentInfo{com.qualcomm.ftcrobotcontroller/org.firstinspires.ftc.robotcontroller.internal.FtcRobotControllerActivity}: android.content.res.Resources$NotFoundException: String resource ID #0x0
        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2690)
        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2755)
        at android.app.ActivityThread.-wrap12(ActivityThread.java)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1495)
        at android.os.Handler.dispatchMessage(Handler.java:102)
        at android.os.Looper.loop(Looper.java:154)
        at android.app.ActivityThread.main(ActivityThread.java:6196)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:888)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:778)
     Caused by: android.content.res.Resources$NotFoundException: String resource ID #0x0
        at android.content.res.Resources.getText(Resources.java:335)
        at android.content.res.Resources.getString(Resources.java:381)
        at android.content.Context.getString(Context.java:476)
        at com.qualcomm.robotcore.util.ClassUtil.decodeStringRes(ClassUtil.java:270)
        at com.qualcomm.robotcore.hardware.configuration.typecontainers.UserConfigurationType.processAnnotation(UserConfigurationType.java:103)
        at com.qualcomm.robotcore.hardware.configuration.ConfigurationTypeManager.filterClass(ConfigurationTypeManager.java:439)
        at org.firstinspires.ftc.robotcore.internal.opmode.ClassManager.processAllClasses(ClassManager.java:263)
        at com.qualcomm.ftccommon.ClassManagerFactory.processAllClasses(ClassManagerFactory.java:92)
        at org.firstinspires.ftc.robotcontroller.internal.FtcRobotControllerActivity.onCreate(FtcRobotControllerActivity.java:330)
        at android.app.Activity.performCreate(Activity.java:6698)
        at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1140)
        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2643)
```

technova12611, 2020-10-08:
Just tried Skystone 5.5 RC app. it worked fine.

closed 2020-10-08 by technova12611

FROGbots-4634, 2020-10-08:
Sorry, why did you close this? If it works with 5.5 but crashes on 6.0, then there's clearly an issue here.

technova12611, 2020-10-09:
I think it's our code issue. After we fixed non-existing string, `@DeviceProperties(xmlTag="RevRobotics1HDHexMotor", name="@string/rev_1_hd_hex_name", builtIn = true)`, it is working fine. The weird thing was that even after we uninstalled our app, and installed the stocked RC app, it's still crashing.