FIRST-Tech-Challenge/FtcRobotController #25: Clarification on the Gamepad Class

JAKCODING, 2020-10-09:
Hi devs,
I'm a programmer on Team 7149. Recently, we've been doing a deep dive into the FTC SDK and we were wondering how the Gamepad class updates during loops. We noticed that in EventLoopManager, the gamepad.fromByteArray() method is called instead of one of the update() functions. How does the fromByteArray method work? How did you determine what hex values to use? Finally, would it be possible to add custom controller support for offseason projects?

Thanks,
Team 7149

NoahAndrews, 2020-10-09:
There's an accurate response here that covers why `fromByteArray` is used instead of `update`: https://github.com/FIRST-Tech-Challenge/SkyStone/issues/275#issuecomment-706251402

> How did you determine what hex values to use?

The `buttons` integer in the `fromByteArray` method contains all of the button data, by using one bit for each of the buttons. The hex values allow individual bits to be retrieved from that integer by using bitwise math. The binary representation of all of the hex values used have only a single `1` digit, with all of the rest being zeros. If you want to get the second-to-last binary digit, you can do a bitwise AND with the hex value corresponding to `0000000000000010`, which is `0x00002`. Hopefully that makes sense, or at least gives you some jumping off points to do more research. With bitwise math, it helps can help to write out the values in binary to see how things work visually.

> Finally, would it be possible to add custom controller support for offseason projects?
Not really, since the Driver Station app would have to have support for it.

JAKCODING, 2020-10-09:
Are there any community projects that allow for modification of the Driver Station app? I'm just wondering, I have a couple ideas for side projects  :)

FROGbots-4634, 2020-10-09:
> Are there any community projects that allow for modification of the Driver Station app? I'm just wondering, I have a couple ideas for side projects  :)

A couple years ago I reverse engineered v3.7 of the DS app. You can find it in the old 2017-18 OpenRC repo.

closed 2021-01-15 by NoahAndrews