FIRST-Tech-Challenge/FtcRobotController #26: error: incompatible types: org.firstinspires.ftc.robotcore.internal.camera.calibration.CameraCalibration cannot be converted to com.vuforia.CameraCalibration

IronReignRobotics, 2020-10-10:
vuforia.getCameraCalibration() no longer returns a native com.vuforia CameraCalibration object. Instead it returns org.firstinspires.ftc.robotcore.internal.camera.calibration.CameraCalibration. I think this first happened in the refactoring in 5.5

There doesn't seem to be a method to create the native object from the new one. This then breaks our ability to use the com.vuforia.Tool class which has the valuable projectPoint methods for calculating offsets from vision targets. This JNI function needs the native vuforia CameraCalibration object.

NoahAndrews, 2020-10-10:
See this issue https://github.com/FIRST-Tech-Challenge/SkyStone/issues/269

IronReignRobotics, 2020-10-11:
Thank you. That was helpful.

NoahAndrews, 2021-01-08:
At this point, we're not going to undo this change, and searching the error message on Google will bring you to that SkyStone issue, so this can be closed.

closed 2021-02-23 by NoahAndrews