FIRST-Tech-Challenge/FtcRobotController #28: WiFi 5Ghz Radio Button always selected

jeroen-umich-edu, 2020-10-14:
After switching to 2.4Ghz Radiobutton, picking a WiFi Channel and then doing 'Apply WiFi Settings' the screen reloads and rather then reflecting how the Control Hub is now configured reverts to selecting the 5Ghz button.    The WiFi network name / control hub name reflect the applied changes and/or current name, but the radio button always reverts back to 5Ghz.    

Should reflect the on-hub settings when it loads as to not accidently change this if other changes are being made (name and/or password)

NoahAndrews, 2020-10-14:
The upcoming 1.1.2 version of the Control Hub OS will allow the Robot Controller app to determine what the current channel is. Until then, the Robot Controller app has no way of knowing what the current channel is, so it selects the 5 GHz band, and doesn't select a specific channel.

NoahAndrews, 2021-01-15:
Control Hub OS 1.1.2 is out, closing this issue.

closed 2021-01-15 by NoahAndrews