FIRST-Tech-Challenge/FtcRobotController #29: Out-of-date links in wiki

Chris857, 2020-10-15:
On https://github.com/FIRST-Tech-Challenge/FtcRobotController/wiki/Downloading-the-Android-Studio-Project-Folder, it says the current season repo is at https://github.com/FIRST-Tech-Challenge/SkyStone when it is actually https://github.com/FIRST-Tech-Challenge/FtcRobotController.

shadope, 2020-12-12:
Can confirm, wasted a bunch of time with this.  Some people, particularly those with some experience, go straight to the Wiki.  Please fix ASAP, or maybe open source the docs?

JIceberg, 2020-12-13:
The wiki is currently severely out of date. If you have suggestions for replacements or want to fix something, I hear they're always willing to accept pull requests. Currently I am working on rewriting the wiki entirely this winter.

JIceberg, 2020-12-13:
> Can confirm, wasted a bunch of time with this.  Some people, particularly those with some experience, go straight to the Wiki.  Please fix ASAP, or maybe open source the docs?  

What do you mean by open source the docs? You can find the javadocs in the README, and the source code for the SDK can be found in Extracted-RC from OpenFTC.

ftctechnh, 2020-12-13:
@Chris857 Thank you for bringing this to our attention.  That wiki page has been corrected with the updated FtcRobotController repo information.

closed 2020-12-13 by ftctechnh

AlecHub, 2020-12-14:
> @Chris857 Thank you for bringing this to our attention. That wiki page has been corrected with the updated FtcRobotController repo information.

Hi Tom, you don't need to use absolute URLs when linking to a page or resource in the repo, from anywhere else in the repo.  GitHub Markdown allows you to create a link to another page or resource by specifying the relative path to the other page or resource.

Use `../` to navigate up one level in the repo, or `../../` to navigate up two levels, and so on.  Then use `folder1/folder2/` to navigate down to the page or resource, as applicable.

For example, to link to the root of the repo from this issue, you would use: 
```
 [Root of this repo from this issue](../)
```
which renders to: [Root of this repo from this issue](../)

To link to a page in the Wiki from this issue, you would use:
```
 [FTC Control System](../wiki/The-FTC-Control-System)
```
which renders to: [FTC Control System](../wiki/The-FTC-Control-System)

To link to a file or resource in a branch of the source tree, say `AndroidManifest.xml` in `master`, you would use:
```
 [Android Manifest](../tree/master/TeamCode/src/main/AndroidManifest.xml)
```
which renders to:  [Android Manifest](../tree/master/TeamCode/src/main/AndroidManifest.xml)

To link to a file or resource in a branch from another file or resource in the same branch, most of the time you would only want to navigate as far up as the root of the branch; not the root of the repo.

If the file or resource is in the same folder, or in a sub-folder, then you don't have to navigate up any levels:
```
 [File1.md in same folder](File1.md)
 [File2.pdf in sub-folder](sub-folder/File2.pdf)
```
`./` would also work, i.e.: `./File1.md` and `./sub-folder/File2.pdf`