FIRST-Tech-Challenge/FtcRobotController #37: Set `RUN_TO_POSITION` target to the current position

NoahBres, 2020-10-21:
When entering RTP, if no target is set, set the target to the position the motor is currently at when the method is called. This way, you don't need to throw the error where target isn't set before RTP and it wouldn't have any odd behavior as it would simply stay at the position it currently is in.

In addition to that, there is a legitimate bug where the error persists even after one uploads the correct code.

JIceberg, 2020-10-21:
RTP as it is right now is kind of janky, especially with that bug. The `TargetPositionNotSetException` is slightly off-putting. I would recommend internally checking to see if the user has ever set a target position before the runmode was set to `RUN_TO_POSITION`. If the user hasn't, then set the target position to the current position so that it doesn't move if some form of `setPower()` is called. Once the user does set a target position, update the target.

In pseudocode it might look something like this:
```python
has_target_set = False

if not has_target_set:
  target_pos = get_current_pos()

def set_target_pos(val):
  has_target_set = True
  target_pos = val
```
An alternative could be to fix that bug where the/an exception persists even when the robot has been shut off and turned back on. I think this might have something to do with the LynxNack standard reason code 50 and the way the exception is specifically handled, but I'm not too sure.

NoahBres, 2020-10-21:
The exception doesn't persist when the entire rev hub has been turned off. It persists between app launches though. If you were to upload code that threw that exception, closed the app, and then uploaded working code, the error will persist. Power cycling the rev hub fixes the issue.

JIceberg, 2020-10-21:
I was referring to this error report:
> every time we turn the robot completely off and then back on again, we get an error for a motor

Perhaps they meant `Restart Robot` instead of "completely off?"

FROGbots-4634, 2020-10-21:
> The exception doesn't persist when the entire rev hub has been turned off. It persists between app launches though. If you were to upload code that threw that exception, closed the app (once the exception is thrown), and then uploaded working code, the error will persist. Power cycling the rev hub fixes the issue.

Hmm, what do you mean by working code? Can you provide 2 example opmodes?

NoahBres, 2020-10-24:
So I cannot figure out how to reproduce that error. My personal encounter was very early in the season so I tried downgrading both the SDK and firmware and it did not suffice. I've seen two other identical reports in the FTC discord server but I have no idea what their configuration was.

Although, one inconsistent behavior is that running a new opmode doesn't seem to clear the target position. So if you run working code(setting target position then RTP, opmode 1), then fire up an opmode (opmode 2) that shouldn't work (setting RTP then target position), the opmode 2 still works and doesn't throw an error. Opmode 2 breaks if you don't run opmode 1 first. 

Even though I cannot reproduce the original issue mentioned, I do still think the default target suggestion would rid of the need for the exception in the first place and make the API more user friendly.

FROGbots-4634, 2020-10-24:
> Although, one inconsistent behavior is that running a new opmode doesn't seem to clear the target position. 

That sounds like a possible firmware bug... Since looking at the StopRobot opmode, (which is what is run when no user opmode is running) it continually puts the RevHub into failsafe mode, which one would think would clear everything in the firmware.

JIceberg, 2020-10-28:
I was able to reproduce the exact same thing as Noah. Unfortunately I couldn't get it to do what we were talking about originally. Here are the [OpModes](https://github.com/JIceberg/FTCLib-Dependency-Tests/tree/master/TeamCode/src/main/java/org/firstinspires/ftc/teamcode/RTPBug) I was working with. When I ran `TestRTPNotWorkingOpMode.java`, I had to restart the robot (press the `Restart Robot` button). This then allowed `TestRTPWorkingOpMode.java` to work. When I ran `TestRTPWorkingOpMode.java` first, it worked as intended. Then, when I ran `TestRTPNotWorkingOpMode.java`, it also worked (as in, no exception was thrown).

JIceberg, 2020-11-14:
Okay so we are now having an issue with the error showing up on code that should work.
```Java
motor.setMode(DcMotor.RunMode.STOP_AND_RESET_ENCODER);

motor.setTargetPosition(5000);

motor.setMode(DcMotor.RunMode.RUN_TO_POSITION);
```
This throws the "must set target position before calling RUN_TO_POSITION." The error persists even after power cycling, resetting the phones, reuploading the code, breaking and fixing the code. I don't know if this is a single test case, but this just happened to a team I mentor.

cmacfarl, 2021-09-25:
The possible firmware bug is being tracked internally.

As to setting the target position to the current position if no target position is set and the run mode is RTP, this is a topic that comes up occasionally with the tech team.   There are two approaches here.  1) What's being advocated for, which I'll call fail silently but gracefully, and 2) the existing implementation, fail fast and loud.   There are legitimate arguments for both approaches.  However as a mentor I've seen teams struggle for extended periods of time when their robot does _nothing_ and they think it should be doing _something_ and they can't figure out why it's doing _nothing_ because there's no user feedback.   Both approaches are, in the end, failures in software.  Only one approach tries to alert the user as to _why_ the failure is happening.   The other approach, especially among inexperienced teams, will produce head scratching.   The decision has been, over the years, to try to minimize head scratching at the expense of a more intrusive failure.  And so, at present, we have fail fast, fail loud.

closed 2021-09-25 by cmacfarl