FIRST-Tech-Challenge/FtcRobotController #38: Tensorflow and Vuforia object detection

Omgkingmoneyman, 2020-10-22:
Our team's programming has been looking through the files and example codes for detecting the Orange rings. Is it a carry over from last years sky stone? We noticed that all the example's are labeled with stones and chips and the such. When using the phone to detect the orange rings it says there are sky stones. Will this be changed/updated in a future release? As well as the information on if there is names for the rings we couldn't find said or commented in the code.

JIceberg, 2020-10-22:
The current game's object detection via TFOD tests for `Quad` and `Single` labels. https://github.com/OpenFTC/OpenRC-Turbo/blob/master/RobotCore/src/main/java/org/firstinspires/ftc/robotcore/external/tfod/TfodCurrentGame.java. The Vuforia for this game is here https://github.com/OpenFTC/OpenRC-Turbo/blob/master/RobotCore/src/main/java/org/firstinspires/ftc/robotcore/external/navigation/VuforiaCurrentGame.java. It is not the same as the skystone models. There are also new samples in the SDK:
* https://github.com/FIRST-Tech-Challenge/FtcRobotController/blob/master/FtcRobotController/src/main/java/org/firstinspires/ftc/robotcontroller/external/samples/ConceptTensorFlowObjectDetection.java
* https://github.com/FIRST-Tech-Challenge/FtcRobotController/blob/master/FtcRobotController/src/main/java/org/firstinspires/ftc/robotcontroller/external/samples/ConceptVuforiaUltimateGoalNavigation.java

NoahAndrews, 2020-10-23:
Make sure you are using version 6.0 of the Robot Controller app , or if you are using Android Studio, that your project is updated to version 6.0.

closed 2020-11-07 by Omgkingmoneyman