FIRST-Tech-Challenge/FtcRobotController #42: App can't launch when deployed via Android Studio

NoahBres, 2020-11-10:
Whenever you click the run button on Android Studio and the app compiles/deploys, the app will launch on the phone and then crash. Launching manually via the home screen or via adb afterwards works.

Here is the logcat for the crashing activity: https://gist.github.com/NoahBres/8994435f1404701d2adb2ada4115c3f9

Error in question:

[image]

FROGbots-4634, 2020-11-10:
I've seen assertion errors on launch for quite some time now (definitely present last season and probably even before) but the app still runs fine. Your log does not show any `FATAL ERROR` messages from the Android Runtime, so as far as I can tell the app still launched normally.

NoahBres, 2020-11-10:
Oh well I guess that error log does not pertain to the app launching problem. It was the only red block of text in logcat so I just assumed that was it.

However, it does 100% not launch properly. Android studio deploys the app. The app starts on the phone. It seems to launch an activity. Then it crashes/exits. If I launch the app again it works fine. 

There were a few others on the discord reporting the same thing.

FROGbots-4634, 2020-11-10:
I have also seen times where deploying from Android Studio does not launch the app properly. However I'm personally of the opinion it might be some sort of bug or race condition within AS itself. At least for me, it happens very rarely.

NoahBres, 2020-11-11:
For me the behavior is seen like 80-90% of the time. Probably just an Android Studio bug I guess ¯\_(ツ)_/¯

Windwoes, 2021-01-06:
> I have also seen times where deploying from Android Studio does not launch the app properly. However I'm personally of the opinion it might be some sort of bug or race condition within AS itself. At least for me, it happens very rarely.

FWIW, the other day I saw this happen multiple times when working on an app completely separate from the FTC software. It's got to be an Android Studio bug.

NoahBres, 2021-01-06:
I noticed this in a a blank test app today. Closing issue.

closed 2021-01-06 by NoahBres