FIRST-Tech-Challenge/FtcRobotController #54: Wrapper for DcMotor/DcMotorEx Causes Controller Activity to Crash On Setting the Power

JIceberg, 2020-11-28:
I'm going to leave an initial issue here because I am genuinely lost. I will leave a link to the actual source code.

Firstly, I have a [`Motor`](https://github.com/FTCLib/FTCLib/blob/dev/core/src/main/java/com/arcrobotics/ftclib/hardware/motors/Motor.java) wrapper class for the `DcMotor` object. The basic structure (related to this issue) is this:
```java
public DcMotor motor;
public Motor(HardwareMap hMap, String name) {
    motor = hMap.get(DcMotor.class, name);
}
public void set(double output) {
    motor.setPower(output);
}
```
Second, I have a [`MotorEx`](https://github.com/FTCLib/FTCLib/blob/dev/core/src/main/java/com/arcrobotics/ftclib/hardware/motors/MotorEx.java) wrapper for the `DcMotorEx` object that extends the `Motor` class. The basic structure is this:
```java
public DcMotorEx motorEx;
public MotorEx(HardwareMap hMap, String name) {
    super(hMap, name);
    motorEx = (DcMotorEx)hMap.get(DcMotor.class, name);
}
@Override
public void set(double output) {
    motorEx.setPower(output);
}
public void setVelocity(double angularRate) {
    motorEx.setVelocity(angularRate);
}
```
In a couple cases I have seen, when `MotorEx.set(double)` is called, it instantly crashes the controller activity. The error it produces in the logcat is that the packets were not returned from the motor (timeout after 250 ms). I do not have a copy of the log files currently. Calling `MotorEx.setVelocity(double)`, however, does not produce this issue.

My question is: what exactly could be causing this issue?

FROGbots-4634, 2020-11-28:
A log file would definitely be helpful

JIceberg, 2020-11-28:
Yeah unfortunately I don't have one with me. I will try to retrieve one ASAP.

JIceberg, 2020-11-29:
[image]
I'm actually shaking and crying right now. Stack overflow from infinite recursion. Issue resolved.

closed 2020-11-29 by JIceberg