FIRST-Tech-Challenge/FtcRobotController #59: Driver Station Disconnects from Control Hub

PriyaSoneji, 2020-12-10:
Our autonomous mode recently started crashing randomly and quite frequently. Looking at the logs we see that the driver station disconnects from the control hub and the op mode crashes. The log is below. We can replicate the issue with several driver station phones of different models such as Moto G Play (Android 7.1.1 and 6.0.1).
Driver Station Version 6.0
Robot Controller Version 6.0
Control Hub OS Version 1.1.1
Control Hub Hardware Client Version 1.0.0
Expansion Hub Version 1.8.2

**Error Log:**
12-07 19:12:10.142   995  1202 V Robocol : sending CMD_NOTIFY_RUN_OP_MODE(1564), attempt: 0
12-07 19:12:18.596   995  1073 D CameraManager: ---------------------------------------------- ACTION_USB_DEVICE_DETACHED
12-07 19:12:18.600   995   995 V FtDeviceManager: ACTION_USB_DEVICE_DETACHED: /dev/bus/usb/003/004
12-07 19:12:18.625   995  1225 E Robocol : exception IOException(Invalid argument): exception sending datagram [java.net.PlainDatagramSocketImpl.send(Native Method)]
12-07 19:12:18.671   995  1201 E Robocol : exception SocketTimeoutException(Receive timed out): no packet received [java.net.PlainDatagramSocketImpl.receive0(Native Method)]
12-07 19:12:18.692   995  1073 D CameraManager: ---------------------------------------------- ACTION_USB_DEVICE_DETACHED
12-07 19:12:18.696   995   995 V FtDeviceManager: ACTION_USB_DEVICE_DETACHED: /dev/bus/usb/003/003
12-07 19:12:18.698   995  1225 D RobotCore: system telemetry: key=$System$Warning$ msg="unable to find webcam "wcam" with serial number 813FCB10"
12-07 19:12:18.700   995  1225 V RobotCore: event loop: device has shutdown abnormally: ABNORMAL
12-07 19:12:18.700   995  1225 V EventLoopManager: event loop: detaching device 813FCB10
12-07 19:12:18.700   995  1225 I FtcEventLoop: vv===== MODULE DETACH RECOVERY: disarm Webcam [813FCB10]=====vv
12-07 19:12:18.701   995  1225 I FtcEventLoop: ======= MODULE DETACH RECOVERY: pretend Webcam [813FCB10]=======
12-07 19:12:18.701   995  1225 I FtcEventLoop: ^^===== MODULE DETACH RECOVERY: complete Webcam [813FCB10]=====^^
12-07 19:12:18.814   995  1205 V SoundInfo: construct(0x0a37d44b)
12-07 19:12:18.856   995  1204 I MediaCodec: callingProcessName:com.qualcomm.ftcrobotcontroller
12-07 19:12:18.860   995  1361 I ACodec  : onAllocateComponent:6703 mSoftCodecPref:0 componentName:
12-07 19:12:18.861   995  1361 I ACodec  : onAllocateComponent:6734 mSoftCodecPref:0 componentName:OMX.google.raw.decoder
12-07 19:12:18.917   995  1073 D SoundPlayer: onLoadComplete(samp=3|ms=2509, samp=3)=0
12-07 19:12:18.920   995  1205 W AudioTrack: AUDIO_OUTPUT_FLAG_FAST denied by client; transfer 4, track 48000 Hz, output 44100 Hz
12-07 19:12:18.926   995  1205 D SoundPlayer: playing volume=0.000000 samp=3|ms=2509
12-07 19:12:18.943   995  1202 V Robocol : sending CMD_PLAY_SOUND(1666), attempt: 0
12-07 19:12:19.183   995  1202 V Robocol : sending CMD_PLAY_SOUND(1666), attempt: 1
12-07 19:12:19.244   995  1271 V RevBlinkinLedDriver: Pattern: BLACK, 99, 0.747500
12-07 19:12:19.342   995  1202 V Robocol : Giving up on command CMD_PLAY_SOUND(1666) after 2 attempts
12-07 19:12:20.382   995  1202 V UpdateUI: Network: active, disconnected
12-07 19:12:20.383   995  1202 I RobotCore: ******************** STOP - OPMODE /storage/emulated/0/FIRST/matchlogs/Match-0-Red-1.txt ********************
12-07 19:12:20.393   995  1202 I RobotCore: Lost connection while running op mode: Red-1
12-07 19:12:20.393   995  1202 V NetworkConnectionHandler: Peer connection lost

cmacfarl, 2020-12-10:
If you take your webcam out of the picture does the problem still occur?

What model of webcam are you using?

PriyaSoneji, 2020-12-10:
We have not tested autonomous without webcam. Will try soon and get back to you. We are using the Logitech C310.

FROGbots-4634, 2020-12-10:
If the Driver Station and Control Hub are connected, and you allow them to sit idling for say 5 minutes, do you observe intermittent disconnects?

PriyaSoneji, 2020-12-10:
We do not observe intermittent disconnects. Disconnects only occur when op mode is running.

FROGbots-4634, 2020-12-10:
> Disconnects only occur when op mode is running.

Can you check if running the following OpMode produces disconnects?

```java
@TeleOp
public class FooBar extends LinearOpMode
{
    int i;
    
    @Override
    public void runOpMode()
    {
        telemetry.setMsTransmissionInterval(20);
        
        waitForStart();
        
        while (opModeIsActive())
        {
            telemetry.addData("Loop count", i);
            telemetry.update();
            sleep(5);
        }
    }
}
```

closed 2021-09-25 by Windwoes