FIRST-Tech-Challenge/FtcRobotController #61: No 5ghz available for direct wifi pairing

daven78, 2020-12-11:
We had been running our robot controller with version 5 from last season until tonight.  We are in a building that has lots of 2.4ghz wifi connections, so we switched to the 5ghz channel and have been running with no issues.  Since we upgraded to 6.0 robot controller it defaults to a 2.4ghz channel and we can't find any way to connect with 5ghz, its not an option!!!!!  Why? Please help, had to go back to the older robot controller version 5 so that we can use the 5ghz channels.

FROGbots-4634, 2020-12-11:
What phones are you using?

daven78, 2020-12-11:
Moto g4's.  We have two rev expansion hubs with the latest firmware.

daven78, 2020-12-11:
> What phones are you using?
Moto g4's. We have two rev expansion hubs with the latest firmware. We are using blocks if that makes a difference.

cmacfarl, 2020-12-11:
When you go to the "Change Wifi Direct Channel" activity under "Advanced Settings" do you not see 5Ghz channels?  e.g.  These screenshots are from a Robot Controller running on an E4 running v6.0

[image]
[image]
[image]

daven78, 2020-12-11:
No, only the 2.4ghz channels show up.  (1 through 11).  When we ran the older 5.0 version they showed up just like your pictures.  We can't figure out why, I thought maybe something changed in the new app or something.

cmacfarl, 2020-12-11:
Oops, my bad.  I thought you said E4.

NoahAndrews, 2020-12-11:
@daven78 Do you have the Moto G4 or the Moto G4 Play? The Moto G4 is an unsupported/not-legal phone, and the Moto G4 Play does not have the hardware for 5 GHz WiFi.

daven78, 2020-12-11:
Oh, I think it's a G4 Play.  I think we have some E4's also.

FROGbots-4634, 2020-12-11:
If it's a G4 Play that begs the question of how were you able to use 5GHz before, if it physically does not have a 5GHz radio?

cmacfarl, 2020-12-11:
Perhaps confusion between G4 and E4 phones?  As I demonstrated earlier today that's pretty easy to do. :-)

daven78, 2020-12-11:
I will have to check with our programmer.  When I upgraded to 6.0, he may have given me the older phones instead of the ones we were already using.  If that's all it is, that will be easy to fix.  I will check it out and let you guys know.

daven78, 2020-12-11:
OK, it's the same phones we were using with the 5.? older robot controller.  When we had trouble with the interference of all the wifi in our building we change the channel to the highest one in the 2.4ghz settings :) not the 5ghz band.  That worked well up until we switched to 6.0.  I could have swore it let us use channel 14 on the 2.4 ghz band, but I just tried to repeat that on the g4 plays and could not.  I reinstalled 5,5.1,5.2,5.3,5.4 and 5.5 and could only access channels 1-11.

Anyway, we have a set of E4's and I put 6.0 on them and set them to the top most 5 ghz channel... yahoo!!!! and the interference is minimal, we have pings of 3ms to 20ms.  On the 2.4ghz band we couldn't get it to go below 400ms or just die with "No Heartbeat".  That's not a pretty picture, the robot controller just continuously resets itself.

Thanks for all your input.  I wasn't looking forward to setting up a faraday cage and look like the battle bots on TV. :)
Dave
Team 9978

cmacfarl, 2020-12-11:
We've never supported channel 14 on 2.4Ghz.   

It appears this was a phone confusion problem.  Closing.

closed 2020-12-11 by cmacfarl