FIRST-Tech-Challenge/FtcRobotController #63: I2CDeviceSynch for serial communications

Team14423, 2020-12-20:
Hi -

We are trying to implement PixyCam2, which uses I2C, but does not use registers. Is there a parameter we can send in for ireg that will cause read() and write() to behave like serial drivers, simply reading data off the port as it comes rather than listing a specific register?

If not, is this why people say to use LegoI2C mode? If so, do you know the register we use for that?

FROGbots-4634, 2020-12-21:
Duplicate of https://github.com/ftctechnh/ftc_app/issues/577

Team14423, 2020-12-21:
Awwww, I got so excited thinking it was fixed. So, there's no solution on this point?

Windwoes, 2021-08-15:
This will be fixed in the next version of the SDK.

Team14423, 2021-08-15:
Awesome - thank you!

Windwoes, 2021-09-20:
Fixed in v7.0

closed 2021-09-20 by Windwoes