FIRST-Tech-Challenge/FtcRobotController #71: Cannot open OpModes in v6.1 Blocks offline editor

lizlooney, 2020-12-25:


texasdiaz, 2021-02-16:
Can verify this behavior. I cannot open existing OpModes nor can I create new OpModes using the v6.1 Blocks Offline Editor. When I attempt it, the page looks like it attempts to load and then fails to a white block diagram (empty page other than the Blocks menu headers).

NoahAndrews, 2021-02-16:
This has been fixed for the next release.

texasdiaz, 2021-02-17:
Any idea when the "next release" will happen? Unfortunately the vast majority of my programming team is "remote" without hardware, and by upgrading everything we have to 6.1 (before we realized this was an issue) we're pretty much dead in the water on the development front.

NoahAndrews, 2021-02-17:
@texasdiaz I don't speak for FTC, so I can't give a specific date, it should be quite soon.

More helpfully, I have a workaround for the meantime:

1. Open the `blocks/FtcBlocks_common.js` file from the unzipped offline editor in a text editor like [Notepad++](https://notepad-plus-plus.org/downloads/).
2. Delete the line of code that contains `Blockly.prompt = parent.showPrompt;`.

See example here: 
[image]

NoahAndrews, 2021-02-23:
Version 6.2 is out now, and fixes this issue.

closed 2021-02-23 by NoahAndrews