FIRST-Tech-Challenge/FtcRobotController #72: Missing doc/apk folder

sbhatnag02, 2020-12-26:
From the ReadMe -- "The folder "apk" contains the .apk files for the FTC Driver Station and FTC Robot Controller apps"

The GitHub repository does not have the doc/apk folder. Because the folder is not there, I cannot find the v6.1 Driver Station App APK. The APK needs to be uploaded, since v6.1 of the Driver Station is not available for download on the Google Play Store.

Windwoes, 2020-12-26:
The APKs are available on the Releases page: https://github.com/FIRST-Tech-Challenge/FtcRobotController/releases

cmacfarl, 2020-12-27:
The APKs are no longer included in the repository itself in an effort to keep the repository size manageable over time.

As @Windwoes mentions, they are on the releases page.  Thank you for pointing out the legacy cruft in the readme, that will be fixed with the next release.

closed 2020-12-27 by cmacfarl