FIRST-Tech-Challenge/FtcRobotController #86: RC breaks WiFi Direction connection with computer on Android Studio upload

NoahBres, 2021-01-12:
Around 1 in every 5 wireless deploys from Android Studio (1 in 5 is just a guesstimate, I haven't actually measured the frequency) will break the WiFi direct connection between my computer and the RC phone. However, the connection between the DS phone and the RC phone is fine.

Here is the log I captured right after my connection died:
[logfile.txt](https://github.com/FIRST-Tech-Challenge/FtcRobotController/files/5803959/logfile.txt)

NoahBres, 2021-01-12:
I will attempt to capture a video of the issue since there's a bit more to it. It's a bit random so will be a second.

Windwoes, 2021-01-12:
Are you certain that this is not an OS issue? (E.g. OS deciding to connect to a different network because the access point broadcast by the phone does not have internet access)?

NoahBres, 2021-01-12:
> Are you certain that this is not an OS issue? (E.g. OS deciding to connect to a different network because the access point broadcast by the phone does not have internet access)?

I have only noticed this issue after 6.1 so I'm 95% sure it's not an OS issue.

NoahBres, 2021-01-12:
https://youtu.be/iTru-u5DpCE

TL;DR:
- I haven't noticed this prior to 6.1. Might be wrong but I downgraded to 6.0 a few days ago, was working on something else for maybe 3 hours, didn't notice it. I am working on something using 6.1 today, experienced it 3 times already.
- I've only noticed it killing the connection between my laptop and RC phone after wirelessly deploying from Android studio, upon the app launching.
- Once the app wirelessly deploys and the app launches, it seems to kill the connection (once every 5ish deploys maybe?). The laptop still thinks it's connected. However, dashboard doesn't connect and `adb connect` doesn't do anything.
- Relaunching the app doesn't fix it
- In that video, I was surprised that simply turning off WiFi and turning it back on fixed it. For my multiple encounters with this issue in the past week, that did not solve it. However, creating the WiFi direct group directly in the WiFi settings rather than letting the app create the WiFi direct group did solve it. That behavior I guess is hit or miss.
    - Just to summarize what I was doing in that case:
        - I encountered the issue. Restarting the app didn't fix it.
        - I would close the app. Turn off WiFi.
        - Turn WiFi back on. Create a WiFi group in the settings.
        - Then launch the app, and it would work.
        - That's what I was going to do in the video but apparently it worked that time fine. In previous attempts, letting the app set up the WiFi direct connection would not solve the issue.

JIceberg, 2021-01-12:
The only thing I see in the logs that might be relevant is the socket closing and the connection being reset for dashboard. I don't see anything else really out of the ordinary? Correct me if I'm wrong.

hartytech, 2021-01-30:
I am having the exact same issue