FIRST-Tech-Challenge/FtcRobotController #90: README.md errors

AlecHub, 2021-01-21:
Reference to the doc/apk and doc/javadoc folders has crept back into README.md.  There are several other reverts in the file.  It appears the current README.md is an edit of the wrong version of the file.

Also, consider putting a link to the FtcDriverStation-release.apk that matches the release in the the release notes of each release, eg:
```
  [v6.1 Driver Station APK](../../releases/download/v6.1/FtcDriverStation-release.apk)
```

Windwoes, 2021-01-21:
> Reference to the doc/apk and doc/javadoc folders has crept back into README.md

Well, as far as the doc/apk goes, #72 indicates that will be fixed in the next release.

AlecHub, 2021-01-21:
> > Reference to the doc/apk and doc/javadoc folders has crept back into README.md
> 
> Well, as far as the doc/apk goes, #72 indicates that will be fixed in the next release.

#72 was closed about a month ago, **with no "next release" in sight**. Issues normally only get closed when the issue is fixed from users' standpoint.  Issues should remain open until fixed.  This will help prevent duplicate issues from being created.

Windwoes, 2021-02-26:
Fixed with 6.2 release

closed 2021-02-26 by Windwoes