FIRST-Tech-Challenge/FtcRobotController #97: setZeroPowerBehavior fails after expansion hub disconnect

danielbrownmsm, 2021-02-06:
Problem: motor on expansion hub will revert to zeroPowerBehavior.BRAKE after losing connection with the control hub then reconnecting. Setting it to zeroPowerBehavior.FLOAT after the disconnect doesn't change this, but also doesn't throw an error.

RC: (Control Hub) OS v. 1.1.2, Android v. 7.1.2, Expansion Hub v. 1.8.2, RC v. 6.1 with an extra expansion hub connected through the RS485 port
DS: moto e5 play android v. 8.0 running DS v. 6.1

the [log](https://github.com/FIRST-Tech-Challenge/FtcRobotController/files/5937347/robotControllerLog.3.txt) is here, with logging low-level events set to true

code can be found [here](https://github.com/danielbrownmsm/FTC-3638-2021) (the AtomicTeleOp, when running the shooter motor in the ShooterSubsystem)

Windwoes, 2021-02-06:
> motor on expansion hub will revert to zeroPowerBehavior.BRAKE after losing connection with the control hub then reconnecting. Setting it to zeroPowerBehavior.FLOAT after the disconnect doesn't change this, but also doesn't throw an error.

Hmm, that's interesting, because I'm 99% sure that FLOAT is the default state on cold power-up.

danielbrownmsm, 2021-02-06:
A possible cause might be using a wrapper class (AtomicMotor) that sets the zeroPowerBehavior to BRAKE in the constructor? I don’t think that would be it, but...
To clarify: I don’t actually know what mode it is set to (I set it to FLOAT, but don’t know if that goes through/registers), but it acts like it is in BRAKE

Windwoes, 2021-02-07:
Interesting. I will see if I can reproduce the issue.

danielbrownmsm, 2021-03-03:
Any progress? I can collect more data if necessary.

Windwoes, 2021-03-04:
Apologies; been busy and haven't gotten to it yet. I hope to test soon though.

Windwoes, 2021-03-09:
@danielbrownmsm I was able to test this, and got some interesting results. There is clearly an issue here, but the behavior I am seeing is actually opposite of what you are reporting. Namely, that if the Expansion Hub is disconnected while an OpMode is live, then the motor controller gets stuck in FLOAT mode. No amount of switching it between BRAKE and FLOAT after the Expansion Hub is reconnected will cause it to enter BRAKE mode again. However, if the OpMode is then stopped and restarted, it will activate BRAKE mode just fine.

As an aside, my testing was also conducted with observations of physical behavior, i.e. turning the motor by hand and feeling whether there was significant resistance (BRAKE) or not (FLOAT).

Windwoes, 2021-03-10:
OK, I see what's going on here. What happens is that when the Expansion Hub enters failsafe mode, part of the failsafe routine is to disable the motor channels. Apparently switching the H-bridge into `FLOAT` mode is part of that disable routine. When the Expansion Hub is re-armed after the connection is restored, the SDK does **not** tell the Expansion Hub to re-enable the motor channels. In this state, the Expansion Hub firmware does still in fact remember that the zero power behavior had been set to `BRAKE` (and commands to change the behavior complete successfully), but the behavior choice is ignored because the motor channel is still disabled.

Either using `motoEx.setMotorEnable()` or simply setting the power of the motor will cause the motor channel to be enabled again, and thus `BRAKE` zero power beahvior will work again.

The question now becomes, should the SDK re-enable the motor channels as part of the Expansion Hub arming process?