FIRST-Tech-Challenge/FtcRobotController #102: Webcams with leading zeros in their serialNumber are not found when loading from res/xml

IronReignRobotics, 2021-02-16:
Webcams with leading zeros in their serialNumber are not found when loading from res/xml. If the config file is stored in the FIRST folder on the Control Hub, the leading zeros are properly recognized and the webcam is found. But when loading from res/xml in the package, a renamed copy of the very same xml file that worked when it was loaded from sdcard/FIRST - it no longer works. If the serial number was "00123456789", the message will say that  Webcam 1 with serial number "123456789" was not found.

Windwoes, 2021-02-16:
This appears to be a [bug](https://issuetracker.google.com/issues/116700688) in the Android resource compiler tool, AAPT2.
This [stack overflow answer](https://stackoverflow.com/a/55582231/) suggests it can be fixed by upgrading the Android Gradle Plugin.

IronReignRobotics, 2021-02-17:
Sweet - that was the fix! Well, I was already on the latest versions of AS/Gradle/Wrapper. But the aapt2 additionalParameters line what what I needed. Inside build.common.gradle of FtcRobotController I added that line to the aaptOptions under the tflite line: 

    aaptOptions {
        noCompress "tflite"
        additionalParameters "--keep-raw-values"
    }

That did the trick. Thanks.

closed 2021-02-17 by IronReignRobotics