FIRST-Tech-Challenge/FtcRobotController #108: Logcat resets when app recovers

IronReignRobotics, 2021-02-26:
Trying to debug a problem with OpenCV we decided to upgrade to 6.2 thinking that might fix the problem. It didn't fix the problem, and now logcat resets when the app auto-recovers. Can we have an advanced setting to disable the watchdog recovery so we can see where the app crashed?

Windwoes, 2021-02-26:
Assuming you are talking about the Control Hub OS level watchdog, the watchdog can be disabled by doing the following:

```
adb shell setprop persist.ch.monitorrc false
```

You can then subsequently re-enable it with

```
adb shell setprop persist.ch.monitorrc true
```

Windwoes, 2021-02-26:
If you are referring to the `stuckInXYZ()` watchdog, note that the watchdog can be disabled by attaching the debugger to the app process.

closed 2021-09-24 by Windwoes