FIRST-Tech-Challenge/FtcRobotController #110: Null error with generic usb webcam

smartynenko, 2021-03-05:
Op mode crashes on attempt to open generic usb camera

```
2021-03-05 11:14:48.720 9200-9334/com.qualcomm.ftcrobotcontroller I/RobotCore: ******************** START - OPMODE Concept: Webcam ********************
2021-03-05 11:14:48.724 9200-9341/com.qualcomm.ftcrobotcontroller I/RobotCore: Attempting to switch to op mode Concept: Webcam
2021-03-05 11:14:48.727 9200-9341/com.qualcomm.ftcrobotcontroller V/ThreadPool: container(0x04a4c337: LinearOpMode) added id=692 TID=0 count=1
2021-03-05 11:14:48.728 9200-9372/com.qualcomm.ftcrobotcontroller V/RobotCore: thread: 'LinearOpMode main' starting...
2021-03-05 11:14:48.729 9200-9372/com.qualcomm.ftcrobotcontroller D/CameraManager: doOpenCamera()...
2021-03-05 11:14:48.730 9200-9372/com.qualcomm.ftcrobotcontroller D/CameraManager: requesting permission for camera: Webcam(3141:25442)
2021-03-05 11:14:48.731 9200-9200/com.qualcomm.ftcrobotcontroller W/com.qualcomm.ftcrobotcontroller: type=1400 audit(0.0:184): avc: denied { search } for uid=10158 comm=4C696E6561724F704D6F6465206D61 name="usb" dev="tmpfs" ino=12498 scontext=u:r:untrusted_app:s0:c512,c768 tcontext=u:object_r:usb_device:s0 tclass=dir permissive=0
2021-03-05 11:14:48.737 9200-9372/com.qualcomm.ftcrobotcontroller D/libusb: [41.418505] [0000249c] libusb: debug [op_create] op_create: szUsbPath=/dev/bus/usb/001/004 szSysfsDir=(null)
2021-03-05 11:14:48.737 9200-9372/com.qualcomm.ftcrobotcontroller D/libusb: [41.418638] [0000249c] libusb: debug [initialize_device] initialize_device(szUsbPath=/dev/bus/usb/001/004 szSysfsDir=(null))
2021-03-05 11:14:48.738 9200-9372/com.qualcomm.ftcrobotcontroller E/libusb: [41.418830] [0000249c] libusb: error [initialize_device] initialize_device(): fdDescriptors unavailable
2021-03-05 11:14:48.738 9200-9372/com.qualcomm.ftcrobotcontroller D/libusb: [41.418869] [0000249c] libusb: debug [_originate_err] [linux_usbfs.c:796] originating err: -12(LIBUSB_ERROR_NOT_SUPPORTED)
2021-03-05 11:14:48.738 9200-9372/com.qualcomm.ftcrobotcontroller V/UvcLibUsbDevice: [jni_libusb_device.cpp:169] libusb_get_serial_number: 
2021-03-05 11:14:48.739 9200-9372/com.qualcomm.ftcrobotcontroller D/libusb: [41.420709] [0000249c] libusb: debug [op_create] op_create: szUsbPath=/dev/bus/usb/001/003 szSysfsDir=(null)
2021-03-05 11:14:48.740 9200-9372/com.qualcomm.ftcrobotcontroller D/libusb: [41.420829] [0000249c] libusb: debug [initialize_device] initialize_device(szUsbPath=/dev/bus/usb/001/003 szSysfsDir=(null))
2021-03-05 11:14:48.740 9200-9372/com.qualcomm.ftcrobotcontroller E/libusb: [41.420975] [0000249c] libusb: error [initialize_device] initialize_device(): fdDescriptors unavailable
2021-03-05 11:14:48.740 9200-9372/com.qualcomm.ftcrobotcontroller D/libusb: [41.421015] [0000249c] libusb: debug [_originate_err] [linux_usbfs.c:796] originating err: -12(LIBUSB_ERROR_NOT_SUPPORTED)
2021-03-05 11:14:48.740 9200-9372/com.qualcomm.ftcrobotcontroller V/UvcLibUsbDevice: [jni_libusb_device.cpp:169] libusb_get_serial_number: 
2021-03-05 11:14:48.742 9200-9372/com.qualcomm.ftcrobotcontroller D/WebcamNameImpl: requesting permission for /dev/bus/usb/001/003
2021-03-05 11:14:48.742 9200-9372/com.qualcomm.ftcrobotcontroller V/WebcamNameImpl: asyncRequestUsbPermission()...
2021-03-05 11:14:48.745 9200-9372/com.qualcomm.ftcrobotcontroller D/WebcamNameImpl: permission already available for /dev/bus/usb/001/003
2021-03-05 11:14:48.745 9200-9355/com.qualcomm.ftcrobotcontroller V/Robocol: sending CMD_NOTIFY_INIT_OP_MODE(466), attempt: 0
2021-03-05 11:14:48.746 9200-9372/com.qualcomm.ftcrobotcontroller V/ThreadPool: container(0x0d80be1b: default threadpool) added id=693 TID=0 count=1
2021-03-05 11:14:48.747 9200-9372/com.qualcomm.ftcrobotcontroller V/WebcamNameImpl: ...asyncRequestUsbPermission()
2021-03-05 11:14:48.748 9200-9373/com.qualcomm.ftcrobotcontroller D/WebcamNameImpl: permission for /dev/bus/usb/001/003=true
2021-03-05 11:14:48.748 9200-9372/com.qualcomm.ftcrobotcontroller D/ContinuationSynchronizer: awaiting(camera open)...
2021-03-05 11:14:48.749 9200-9373/com.qualcomm.ftcrobotcontroller D/CameraManager: permission granted for camera: Webcam(3141:25442)
2021-03-05 11:14:48.750 9200-9373/com.qualcomm.ftcrobotcontroller D/CameraManager: asyncOpenCamera(Webcam(3141:25442))...
2021-03-05 11:14:48.752 9200-9373/com.qualcomm.ftcrobotcontroller D/CameraManager: asyncOpenWebcam(Webcam(3141:25442))...
2021-03-05 11:14:48.752 9200-9373/com.qualcomm.ftcrobotcontroller V/UsbResiliantWebcam: construct(0x00b991c2)
2021-03-05 11:14:48.754 9200-9373/com.qualcomm.ftcrobotcontroller D/UsbResiliantWebcam: openAssumingPermission()...
2021-03-05 11:14:48.758 9200-9373/com.qualcomm.ftcrobotcontroller D/UvcContext: usb device is *not* UVC compatible, /dev/bus/usb/001/004
2021-03-05 11:14:48.759 9200-9373/com.qualcomm.ftcrobotcontroller D/UvcContext: found webcam: usbPath=/dev/bus/usb/001/003 vid=3141 pid=25442 serial=null product=USB 2.0 Camera
2021-03-05 11:14:48.759 9200-9373/com.qualcomm.ftcrobotcontroller D/Uvc: [jni_context.cpp:120] Java_org_firstinspires_ftc_robotcore_internal_camera_libuvc_nativeobject_UvcContext_nativeCreateUvcDevice()...
2021-03-05 11:14:48.759 9200-9373/com.qualcomm.ftcrobotcontroller V/Uvc: API: Java_org_firstinspires_ftc_robotcore_internal_camera_libuvc_nativeobject_UvcContext_nativeCreateUvcDevice: enter...
2021-03-05 11:14:48.760 9200-9373/com.qualcomm.ftcrobotcontroller D/libusb: [41.440757] [0000249d] libusb: debug [op_create] op_create: szUsbPath=/dev/bus/usb/001/003 szSysfsDir=(null)
2021-03-05 11:14:48.760 9200-9373/com.qualcomm.ftcrobotcontroller D/libusb: [41.440823] [0000249d] libusb: debug [initialize_device] initialize_device(szUsbPath=/dev/bus/usb/001/003 szSysfsDir=(null))
2021-03-05 11:14:48.760 9200-9373/com.qualcomm.ftcrobotcontroller E/libusb: [41.441260] [0000249d] libusb: error [initialize_device] initialize_device(): fdDescriptors unavailable
2021-03-05 11:14:48.760 9200-9373/com.qualcomm.ftcrobotcontroller D/libusb: [41.441318] [0000249d] libusb: debug [_originate_err] [linux_usbfs.c:796] originating err: -12(LIBUSB_ERROR_NOT_SUPPORTED)
2021-03-05 11:14:48.760 9200-9373/com.qualcomm.ftcrobotcontroller D/Uvc: [device.cpp:613] uvc_create_uvc_device()...

```

Windwoes, 2021-03-05:
What Robot Controller device, camera model, and SDK version are you using?

smartynenko, 2021-03-05:
Moto e5 play, 6.2

Camera data:

```
[Port1]  :  USB Composite Device

Is Port User Connectable:         yes
Is Port Debug Capable:            no
Companion Port Number:            0
Companion Hub Symbolic Link Name: 
Protocols Supported:
 USB 1.1:                         yes
 USB 2.0:                         yes
 USB 3.0:                         no

Device Power State:               PowerDeviceD0

       ---===>Device Information<===---
English product name: "USB 2.0 Camera"

ConnectionStatus:                  
Current Config Value:              0x01  -> Device Bus Speed: High (is not SuperSpeed or higher capable)
Device Address:                    0x19
Open Pipes:                           1

          ===>Device Descriptor<===
bLength:                           0x12
bDescriptorType:                   0x01
bcdUSB:                          0x0200
bDeviceClass:                      0xEF  -> This is a Multi-interface Function Code Device
bDeviceSubClass:                   0x02  -> This is the Common Class Sub Class
bDeviceProtocol:                   0x01  -> This is the Interface Association Descriptor protocol
bMaxPacketSize0:                   0x40 = (64) Bytes
idVendor:                        0x0C45 = Sonix Technology Co., Ltd.
idProduct:                       0x6362
bcdDevice:                       0x0000
iManufacturer:                     0x02
     English (United States)  "Sonix Technology Co., Ltd."
iProduct:                          0x01
     English (United States)  "USB 2.0 Camera"
iSerialNumber:                     0x00
bNumConfigurations:                0x01

          ---===>Open Pipes<===---

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x83  -> Direction: IN - EndpointID: 3
bmAttributes:                      0x03  -> Interrupt Transfer Type
wMaxPacketSize:                  0x0010 = 1 transactions per microframe, 0x10 max bytes
bInterval:                         0x06

       ---===>Full Configuration Descriptor<===---

          ===>Configuration Descriptor<===
bLength:                           0x09
bDescriptorType:                   0x02
wTotalLength:                    0x0663  -> Validated
bNumInterfaces:                    0x02
bConfigurationValue:               0x01
iConfiguration:                    0x00
bmAttributes:                      0x80  -> Bus Powered
MaxPower:                          0xFA = 500 mA

          ===>IAD Descriptor<===
bLength:                           0x08
bDescriptorType:                   0x0B
bFirstInterface:                   0x00
[... 938 lines cut ...]

          ===>Interface Descriptor<===
bLength:                           0x09
bDescriptorType:                   0x04
bInterfaceNumber:                  0x01
bAlternateSetting:                 0x06
bNumEndpoints:                     0x01
bInterfaceClass:                   0x0E  -> Video Interface Class
bInterfaceSubClass:                0x02  -> Video Streaming Interface SubClass
bInterfaceProtocol:                0x00
iInterface:                        0x00

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x81  -> Direction: IN - EndpointID: 1
bmAttributes:                      0x05  -> Isochronous Transfer Type, Synchronization Type = Asynchronous, Usage Type = Data Endpoint
wMaxPacketSize:                  0x1400 = 3 transactions per microframe, 0x400 max bytes
bInterval:                         0x01

          ===>Additional Error Checking<===
PASS: number of uncompressed-frame frame descriptors (15) == number of frame descriptors (15) specified in uncompressed format descriptor(s)
PASS: number of MJPEG frame descriptors (16) == number of frame descriptors (16) specified in MJPEG format descriptor(s)

```