FIRST-Tech-Challenge/FtcRobotController #111: version mismatch: Version 6.1 FtcRobotController vs Version 6.0 FtcDriverStation

RoboWarriors14657, 2021-03-05:
I have a question about FtcRobotController and FTC driver station version mismatch issue.

Our team downloaded and installed FtcRobotController version 6.1 from the site below:
https://github.com/FIRST-Tech-Challenge/FtcRobotController

We installed a FtcDriverStation latest version 6.0 from the Play Store, when we do the self inspect we got a RED flag about version mismatch: Version 6.1 FtcRobotController vs Version 6.0 FtcDriverStation.

This is just in the self inspect and the robot can still function with the warning message "DS version does not match RC".

I google search online but I could find much information about it.  Are there any other teams report this? Any solutions for this issue?

Thanks,

Joey
[image]

Windwoes, 2021-03-05:
The FTC apps on Google Play are no longer being updated, due to Google recently changing some requirements for apps to be hosted on the Play Store. The v6.1 APKs for the Driver Station and Robot Controller can be found on the releases page: https://github.com/FIRST-Tech-Challenge/FtcRobotController/releases/tag/v6.2

You can also use the REV Hardware Client to install the updated apps.

RoboWarriors14657, 2021-03-05:
I downloaded and install the apk file.  It works now!
Thanks,
Joey

closed 2021-03-05 by Windwoes