FIRST-Tech-Challenge/FtcRobotController #116: Switch to an incremental annotation processor?

Superman132, 2021-03-17:
When compiling my project, I can get a build warning saying that the org.firstinspires.opmodeannotationprocessor.MyProcessor is non-incremental, meaning the entire compiler must run non incrementally. This means that builds can become very slow because it has to rebuild all files every time a change occurs.

(The above is my understanding, but could be very wrong. [Google's guide to improving build speed with annotation processors](https://developer.android.com/studio/build/optimize-your-build?utm_source=android-studio#annotation_processors))

Would it be possible to switch to an incremental annotation processor to speed up builds?

Windwoes, 2021-09-23:
The annotation processor is simply doing compile-time analysis to check for OpMode name clashes; you can disable it without losing any core functionality. Does disabling it make any noticeable difference in build times?

Windwoes, 2024-09-14:
The annotation processor has been removed from the SDK.

closed 2024-09-14 by Windwoes