FIRST-Tech-Challenge/FtcRobotController #119: robot controller6.2 crash in android 10

steveliu168, 2021-03-26:
when I try to use Driver Station to connect robot controller,RC  crash every time. My android version is 10 ,and I try android 9,it has the same issue. But it works perfect in android 8.1.
I notice that the logs has tow orange information: libc :Access denied finding property "ro.serialno"

Windwoes, 2021-03-26:
This is currently a known issue. As no competition legal devices currently run Android 10, fixing this has not been high on the priority list, but it will be investigated at some point.

pennygao, 2021-04-17:
I experience RobotController 6.2 crash on Android 10, too, also as soon as Driver Station connected to Robot Controller. But I don't see "libc :Access denied finding property "ro.serialno" in logs, instead, my crash backtrace is like below:

04-16 18:55:52.641 15797 15797 F DEBUG   : signal 31 (SIGSYS), code -6 (SI_TKILL), fault addr --------

04-16 18:55:53.025 15797 15797 F DEBUG   : backtrace:
04-16 18:55:53.025 15797 15797 F DEBUG   :       #00 pc 00000000000cea28  /apex/com.android.runtime/lib64/bionic/libc.so (settimeofday+8) (BuildId: 4cd97724e0ae0bf213b33eacb7bfbbe8)
04-16 18:55:53.026 15797 15797 F DEBUG   :       #01 pc 0000000000021eb4  /data/app/com.qualcomm.ftcrobotcontroller-G5ASKIaCF3aKTBC4z9uJcg==/lib/arm64/libRobotCore.so (Java_org_firstinspires_ftc_robotcore_internal_system_AppUtil_nativeSetCurrentTimeMillis+156) (BuildId: 6b9c6ad4e62aa663f15cf9cefbfcd8cbce597f33)
04-16 18:55:53.026 15797 15797 F DEBUG   :       #02 pc 000000000002f374  /data/app/com.qualcomm.ftcrobotcontroller-G5ASKIaCF3aKTBC4z9uJcg==/oat/arm64/base.odex (art_jni_trampoline+132)
04-16 18:55:53.026 15797 15797 F DEBUG   :       #03 pc 0000000000137334  /apex/com.android.runtime/lib64/libart.so (art_quick_invoke_stub+548) (BuildId: e1ef08b5ba2ec71adc5b7c8230ae4048)
04-16 18:55:53.026 15797 15797 F DEBUG   :       #04 pc 000000000014606c  /apex/com.android.runtime/lib64/libart.so (art::ArtMethod::Invoke(art::Thread*, unsigned int*, unsigned int, art::JValue*, char const*)+244) (BuildId: e1ef08b5ba2ec71adc5b7c8230ae4048)

With some google search it seems settimeofday() requires root privilege and probably this is why the app get killed. Is this the same problem as reported above, or a different one?

(Complete log collected through "adb logcat" and AndroidStudio attached.

[log_from_adb_logcat.log](https://github.com/FIRST-Tech-Challenge/FtcRobotController/files/6328529/log_from_adb_logcat.log)
[log_from_AndroidStudio.log](https://github.com/FIRST-Tech-Challenge/FtcRobotController/files/6328530/log_from_AndroidStudio.log)

NoahAndrews, 2021-04-17:
Good sleuthing! We already figured out that was the issue, and have it fixed for the next release.

Windwoes, 2021-04-17:
> With some google search it seems settimeofday() requires root privilege and probably this is why the app get killed. Is this the same problem as reported above, or a different one?

Yes, this is indeed the issue. The Control Hub has a custom kernel to allow the system call, and unmodified Android versions before Pie will simply silently fail when making the call. However, newer versions will cause a native crash if an app tries to invoke that system call.

This will be fixed in the next release of the SDK.

Windwoes, 2021-09-20:
Fixed in 7.0

closed 2021-09-20 by Windwoes