FIRST-Tech-Challenge/FtcRobotController #121: Does track pose work in blocks for ultimate goal?

tjhawkey, 2021-04-01:
I'm trying to extract pose information using blocks for a teaching example, but any call of track pose generates this error message
Error: Fatal error occurred while executing the block labeled "call VuforiaUltimateGoal.trackpose"

I took the canned vuforia webacm example and added one block.
Works fine when this block is disabled. Image attached.

[image]

lizlooney, 2021-05-29:
The trackPose block is supposed to:

> Returns the VuforiaTrackingResults of the pose of the trackable with the given name. ' +
        'The pose is the location of the trackable in the phone\'s coordinate system.

It is not useful when using a webcam. It shouldn't crash though!

tjhawkey, 2021-05-29:
Thanks. I don't yet see why track pose wouldn't work with a webcam since you can return global coordinate results from vuforia with a webcam. Doesn't it just go by what camera is defined for vuforia? It would be very nice to get track pose to work in blocks.