FIRST-Tech-Challenge/FtcRobotController #125: Throw an error—with an appropriate/helpful message—if an ID is not found in `hardwareMap`

NoahBres, 2021-04-04:
Users often misconfigure the ID's in their own code (or copy-paste existing code without changing the ID's to match their own  config). The SDK simply returns a `null` value and they do not realize their error until they try to act on that object later on in their code, running into a `NullPointerException`.

Most students who wonder what is going on with their code do not understand what this means; most of them probably aren't even aware of what a `null` value even entails.
This is a very common issue.

I believe it would help a lot if the SDK threw an exception on `hardwareMap.get()` and surfaced an appropriate error message `"Device by the name of [x] was not found in your current hardware config"` or something.

Something neat would be a Levenshtein distance check on the provided ID and the ID's that exist in the config. So it could surface an error like:
`Device by the name of "calw" was not found. Did you mean "claw"?`

Windwoes, 2021-04-04:
> The SDK simply returns a `null` value

Actually, that's not true - it throws an IllegalArgumentException.

Here's an OpMode which demonstrates this:

```java
@TeleOp
public class HwMapTester extends LinearOpMode
{
    DcMotorEx motor;
    Servo servo;

    @Override
    public void runOpMode()
    {
        motor = hardwareMap.get(DcMotorEx.class, "badMotorNameXYZ");
        servo = hardwareMap.servo.get("badServoNameXYZ");

        waitForStart();

        while (opModeIsActive())
        {
            sleep(20);
        }
    }
}
```

NoahBres, 2021-04-04:
Turns out I am stupid. My apologies!

closed 2021-04-04 by NoahBres