FIRST-Tech-Challenge/FtcRobotController #129: Mecanum Drive Sample

dansman805, 2021-04-11:
Adding a mecanum drive sample would be very useful given the proportion of teams using mecanum drive. Given there already is a pushbot tank drive tutorial, a mecanum drive tutorial would fit in with the existing samples. The [mecanum drive tutorial page[(https://gm0.org/en/latest/docs/software/mecanum-drive.html) of gm0 is the most popular (non-index) page on the whole site—it's clearly something the community is in need of. Of course, having integrated drive classes for both differential drive and mecanum drive would be great, but a sample would likely be easier to add.

Windwoes, 2021-12-05:
A mecanum drive sample will be included in the next release of the SDK.

Windwoes, 2022-01-17:
Addressed in 7.1

closed 2022-01-17 by Windwoes