FIRST-Tech-Challenge/FtcRobotController #132: Driver Station app 6.2 does not detect gamepad on Android 10

pennygao, 2021-04-24:
I cannot get Driver Station app 6.2 to detect my gamepad (Logitech F310) on Android 10: there's no gamepad icon on top of "user 1" after I press "Start + A", while I can see this on another android phone running Android 8.1.

By comparing the driverStationLog.txt, I found the one from Android 8.1 has below 2 lines, while the one from Android 10 does not (will attach the full log):

04-23 19:07:39.696  9927  9927 V GamepadManager: Input device 4 has been autodetected based on USB VID and PID as LogitechGamepadF310
04-23 19:07:40.457  9927  9927 V GamepadManager: assigned id=4 user=ONE type=LOGITECH_F310 class=LogitechGamepadF310

Meanwhile, on the Android 10 phone I installed an app called OTG Guru, which is able to show me the gamepad details, to confirm there is no connection problem. I'll also attach the screen shot.

[driverStationLogFromAndroid8.1.txt](https://github.com/FIRST-Tech-Challenge/FtcRobotController/files/6368864/driverStationLogFromAndroid8.1.txt)
[driverStationLogFromAndroid10.txt](https://github.com/FIRST-Tech-Challenge/FtcRobotController/files/6368865/driverStationLogFromAndroid10.txt)
[image]

Windwoes, 2021-04-24:
Does the F310 gamepad work in the Android system UI on the Android 10 device? For instance, can you use it to navigate around the app tray?

If not, then it seems that for some reason your Android 10 device is not correctly registering the gamepad with Android's InputDevice API, which unfortunately not much can be done about that. (The Driver Station currently uses the InputDevice API).

The gamepad showing up in OTG Guru means that it's available for use in the USB host API, but it doesn't provide any indication about the InputDevice API.

pennygao, 2021-04-24:
The gamepad is not working for Android system UI (on Android 10). Looks like an Android problem, then.

Windwoes, 2021-08-15:
The next version of the SDK will contain changes which should workaround this issue.

Windwoes, 2021-09-20:
Should be fixed in v7.0

closed 2021-09-20 by Windwoes