FIRST-Tech-Challenge/FtcRobotController #145: Driver Hub at 6.2.2 gives warning when connected to FtcRobotController built from 6.2

alan412, 2021-08-09:
The Driver Hub from REV is at Driver Station 6.2.2

This builds a robot controller to version 6.2

So the Driver Hub gives a warning that the two don't match.   I am concerned about the students learning to ignore that warning...

Is there any way you could release a robot controller 6.2.2.  (or a version of both of them that is larger)

Thanks in advance,

Alan
Coach FTC #16072, Quantum Quacks

cmacfarl, 2021-08-09:
This is a known issue.  The mismatch is harmless from a functional perspective, but a new APK for the Driver Hub was necessary, and there was no reason to release a new robot controller, so here we sit.  When v7.0 is released for Freight Frenzy this will go away.

alan412, 2021-08-10:
I would humbly suggest that not training students to ignore warning messages is a reason to release a new robot controller. But I understand that with limited resources that might not be enough of a reason.

alan412, 2021-08-10:
@AlecHub - Thanks but since we only have one driver hub then we'll have the same problem when a regular driver station tries to connect to our robot controller since we can't rebuild the driver station....  (except that will be 6.2 trying to connect to 6.2.2)

AlecHub, 2021-08-10:
I suggest that Version **X.Y**.Z1 of RC/DS/DH be compatible with Version **X.Y**.Z2 of RC/DS/DH, and vice-versa, by design.  No version mismatch warnings should be displayed if first two numbers (**X.Y**) of the versions match.

cmacfarl, 2021-08-10:
@AlecHub I agree and thought about that as I composed my initial response.   Filter version mismatch by major.minor, disregard point releases.  We'll put it on the list to discuss with the tech team.

alan412, 2021-08-10:
@AlecHub That sounds like a great suggestion that would solve the problem in the future.

NoahAndrews, 2021-09-20:
7.0 is out now, and implements @AlecHub's suggestion.

closed 2021-09-20 by NoahAndrews