FIRST-Tech-Challenge/FtcRobotController #154: Android Studio Compile of 6.0 controller app fails

Crossle86, 2021-09-17:
There is probably a simple answer for this, but just getting back into FTC after months away. Opened my 6.0 controller project expecting to work with that to get going again while waiting for 2021/2022 release. However, I cannot compile anything as all imports of org.firstinspires.ftc.robotcore packages fail with Cannot resolve symbol 'robotcore'.

NoahAndrews, 2021-09-17:
The 3rd-party hosting service we were using for 6.0 artifacts (called Bintray) was shut down. Update your project to version 6.2, and that will work.

NoahAndrews, 2021-09-20:
7.0 is out now.

closed 2021-09-20 by NoahAndrews

Crossle86, 2021-09-20:
Thanks for the quick response and 7.0 notification.

7.0 is out now.
>
> You are receiving this
because you authored the thread.
view it on GitHub:
https://github.com/FIRST-Tech-Challenge/FtcRobotController/issues/154#issuecomment-923080726

Richard Corn
Software Devices LLC
P.O. Box 12299, Olympia WA. 98508
***@***.***Â Â Â www.swdevs.com