FIRST-Tech-Challenge/FtcRobotController #160: Kotlin support?

baqwas, 2021-09-21:
Hello,

This is **NOT** an issue but rather an inquiry emanating from a self-inflicted choice. I am using Android Studio Arctic Fox (2020.3.1) at a _"Hello World"_ level.

Can I use Kotlin code for planned REV FTC project FIRST FREIGHT?

I stopped teaching Java at a local university (for Continuing Education classes) before Android was launched but I feel that I could help a local team be more productive with Android Studio (especially for object detection with TFL). Sorry for posting my inquiry in this manner but having browsed through several sites (and _read-only_ wiki sites) I could not find the answer to my question. Since I want to use Android Studio with the REV solutions, this note was the easiest recourse for me. Thanks for your understanding.

Kind regards.

cmacfarl, 2021-09-21:
Kotlin is not a requirement for using Android Studio.  All the samples are straight up Java, and the entire set of AAR's that provide the foundation of the SDK are Java.  My recommendation would be go ahead and teach you students Java and program your opmodes in Java.

baqwas, 2021-09-21:
Thanks @cmacfarl for helping me to stay on track. I will follow your advice gladly.

Kind regards.

closed 2021-09-21 by baqwas