FIRST-Tech-Challenge/FtcRobotController #162: build.gradle has wrong version of modules in doc

alan412, 2021-09-21:
Lines 41:45 should be:
```
    doc 'org.firstinspires.ftc:Hardware:7.0.0'
    doc 'org.firstinspires.ftc:RobotCore:7.0.0'
    doc 'org.firstinspires.ftc:FtcCommon:7.0.0'
    doc 'org.firstinspires.ftc:OnBotJava:7.0.0'
    doc 'org.firstinspires.ftc:Inspection:7.0.0'
```
This is needed so extractJavadoc can be run to have the Javadoc locally

cmacfarl, 2021-09-21:
Yeah, that didn't get updated because official hosting of javadocs has migrated to https://javadoc.io/doc/org.firstinspires.ftc and, when late in the release process it was discovered that this was a leftover from prior releases the consensus was anyone experienced enough to know what this is for would know to change those version numbers if they wanted to extract javadoc locally.  Additionally all of the other versioning that is done in gradle is done via the tooling that produces the release.   But there's no existing tooling for updating that particular dependencies block.

We can consider adding that tooling, or perhaps remove that block entirely.   There's a download button on javadoc.io so if you really want local copies you can get them there.  I'm somewhat in favor of simplification over adding more tooling given that an alternative is readily available.

alan412, 2021-09-21:
I think it could be pulled.   You can use Android Studio to create the
javadoc instead of gradle.   I am a big fan of simplifying by removing
unnecessary code.

On a related subject, I would be a strong supporter of simplifying the
gradle scripts to remove redundancy.  For example, while TeamCode uses the
root build.common.gradle, the FtcRobotController does not.

I am willing to help with this cleanup, but not sure what the best way to
do that is.   (I could do pull requests, but if there is a better way then
please let me know.)

--Alan

cmacfarl, 2021-09-21:
That particular block was being used to generate the javadoc that was going up on GitHub pages, which as I mentioned isn't being used anymore since the move the Maven and the free scraping by javadoc.io.

The release process isn't set up for the tech team to be able to take pull requests on the public SDK repository as any changes on FtcRobotController are at risk of getting overwritten on the next release, and the goal, not always met, but the goal anyhow is to have just release commits in the commit history of FtcRobotController in order to keep the commit history as clean as possible.

The gradle in question is shared between the public SDK and the private development SDK and goes through a bit of a transformation as it's pushed to the release and some of what you see in the release is because of dependencies on how things are built in the private SDK and avoiding duplication there.   If you really want to contribute the best course of action is to get know the OpenFTC people.  Maybe find something there needs to be fixed.  The tech team has in the past taken contributions to OpenFTC and pulled them into the official SDK.   If your goal for anything is upstreaming, getting to know the principles there, and getting them onboard with what you want to do is the best path forward.

cmacfarl, 2021-09-24:
This is being tracked internally.  Thank you for you comments.

closed 2021-09-24 by cmacfarl