FIRST-Tech-Challenge/FtcRobotController #163: Broken Java Doc link in README

dansman805, 2021-09-21:
The Java Doc link under [this section](https://github.com/first-tech-challenge/ftcrobotcontroller#javadoc-reference-material) of the README goes to https://javadoc.io/org.firstinspires.ftc, which fails with an error " /org.firstinspires.ftc is not valid". Instead, it should link to https://javadoc.io/doc/org.firstinspires.ftc (note the `/doc` in the url preceding `/org.firstinspires.ftc`).

cmacfarl, 2021-09-21:
Oh good grief.  That makes me so unhappy...

cmacfarl, 2021-09-21:
At least it's correct over here: https://github.com/FIRST-Tech-Challenge/FtcRobotController/releases/tag/v7.0

dansman805, 2021-09-21:
I've now checked the Javadoc links in the wiki as well, and while they work, they are out of date. The Javadoc links [here](https://github.com/FIRST-Tech-Challenge/FtcRobotController/wiki/Using-an-External-Webcam-with-Control-Hub#advanced-users) and [here](https://github.com/FIRST-Tech-Challenge/FtcRobotController/wiki/OnBot-Java-Reference-Info#javadoc-reference-pages) still point to https://first-tech-challenge.github.io/SkyStone. The link [here](https://github.com/FIRST-Tech-Challenge/FtcRobotController/wiki/Creating-and-Running-an-Op-Mode-(Android-Studio)#javadoc-reference-information) points to       http://ftctechnh.github.io/ftc_app/doc/javadoc/index.html.

cmacfarl, 2021-09-21:
> I've now checked the Javadoc links in the wiki as well, and while they work, they are out of date. The Javadoc links [...]

I've fixed these two entries.  The main page readme will be fixed with the next release.

dansman805, 2021-09-22:
Thanks for updating those two entries, however there is still an outdated wiki entry [here](https://github.com/FIRST-Tech-Challenge/FtcRobotController/wiki/Creating-and-Running-an-Op-Mode-(Android-Studio)#javadoc-reference-information).

WestsideRobotics, 2021-09-22:
> Thanks for updating those two entries, however there is still an outdated wiki entry [here](https://github.com/FIRST-Tech-Challenge/FtcRobotController/wiki/Creating-and-Running-an-Op-Mode-(Android-Studio)#javadoc-reference-information).

Appreciate that last one, now fixed.

cmacfarl, 2021-09-23:
The readme fix is in progress for the next release.

closed 2021-09-23 by cmacfarl