FIRST-Tech-Challenge/FtcRobotController #166: TensorFlow Marker detection

ftc14158, 2021-09-22:
Is the new TF model for 'Marker' supposed to detect the tape squares on the floor that represent the Freight Frenzy 'barcode'?

It does not appear to be detecting a blue tape square for us here. What is the model based on?

gearsincorg, 2021-09-24:
That was the original intent, and also to reduce the likelihood that an empty spot would detect a duck.  In the end I'm not sure the tape provides a reasonable object to detect, since it has no  texture or features.

cmacfarl, 2021-09-24:
I'll leave this open for a while so if people come here looking for information on this topic, they'll find it, but further discussion of this topic is probably better over in the forums or reddit.

ftc14158, 2021-09-24:
You can go ahead and close, we did manage to get it to detect a blue marker square briefly. Hopefully it will be more reliable once on the robot proper.

closed 2021-09-24 by cmacfarl