FIRST-Tech-Challenge/FtcRobotController #167: Repository contains random teams' code?

sudoer777, 2021-09-23:
Out of curiosity, I was exploring commits within the repository, and I found commits to what appears to be random teams' code.
These commits are not part of any branch, but they exist in the repository and have not been removed.

I found this when I cloned this repo to a private Gitea server and explored the commit graph:
[image]

Although these commits are not visible in Github or any branch, they are still publicly visible by cloning the repo and exploring a bit. I don't know if this is intentional or not, but it looks like a possible concern.

Windwoes, 2021-09-23:
Are you sure you're not viewing a tree that includes forks somehow?

sudoer777, 2021-09-23:
Actually I just explored some more and apparently Gitea is also pulling code from pull requests for some reason, so it's not an issue with the repo after all.

closed 2021-09-23 by sudoer777