FIRST-Tech-Challenge/FtcRobotController #168: jcenter deprecation

yeSpud, 2021-09-24:
[JCenter is shutting down](https://blog.gradle.org/jcenter-shutdown).

Because maven central is already being used as a repository, jcenter should be removed from [build.dependencies.gradle](https://github.com/FIRST-Tech-Challenge/FtcRobotController/blob/master/build.dependencies.gradle#L4) (as maven central is the goto replacement for jcenter).

This will also stop gradle from emitting deprecation warnings.

yeSpud, 2021-09-24:
Forgot that jcenter is also used in [build.gradle](https://github.com/FIRST-Tech-Challenge/FtcRobotController/blob/master/build.gradle)

cmacfarl, 2021-09-24:
Thank you for the reminder.  This is being tracked internally.   Clarification however, that JFrog has announced that it is leaving JCenter online in a read only fashion indefinitely.  It's not however clear yet whether or not the gradle team will reverse track on their decision to remove the jcenter support in v8.0.

closed 2021-09-24 by cmacfarl