FIRST-Tech-Challenge/FtcRobotController #169: Feature request: Update gradle versions

yeSpud, 2021-09-24:
The Android Gradle Plugin can be upgraded to [version 7](https://developer.android.com/studio/releases/gradle-plugin?buildsystem=ndk-build#7-0-0)(.0.2) to take advantage of new features and bug fixes.

To use the updated Gradle plugin, [Gradle itself will have to be upgraded to version 7](https://developer.android.com/studio/releases/gradle-plugin?buildsystem=ndk-build#updating-gradle)(.2).

If you're worried about the JDK being invalid as Gradle now requires JDK 11, rest assured as Android Studio comes bundled with [JDK 11](https://developer.android.com/studio/releases/gradle-plugin?buildsystem=ndk-build#jdk-11).

*Also since Android Studio comes bundled with JDK 11 can the compile options in [build.gradle](https://github.com/FIRST-Tech-Challenge/FtcRobotController/blob/master/FtcRobotController/build.gradle#L19) and [build.common.gradle](https://github.com/FIRST-Tech-Challenge/FtcRobotController/blob/master/build.common.gradle#L110) be [updated to take advantage of it as well](https://developer.android.com/studio/releases/gradle-plugin?buildsystem=ndk-build#java-11)?*

Windwoes, 2021-09-24:
Is there a particular reason that Gradle 7.0 is desirable beyond the fact that AS suggests to update to it? You're free to change what Gradle version you use for your workspace specifically :)

cmacfarl, 2021-09-24:
As @Windwoes pointed out, users who desire an upgraded plugin may change it in their local clone or fork.

closed 2021-09-24 by cmacfarl