FIRST-Tech-Challenge/FtcRobotController #172: Driver Station v7.0 Crashes on Moto G (3rd gen)

crowcasso, 2021-09-25:
The DS app works until changing the Pairing Method to Control Hub. After that change the app immediately crashes and any subsequent start of the app immediately crashes.

Windwoes, 2021-09-25:
We are not able to reproduce this issue on Moto G 3rd Gen phones.
Are you able to capture logcat output of the crash?

If you use Android Studio, you can do the following:
 1. Connect DS phone to computer
 2. Open the logcat pane in Android Studio
 3. Set the logcat window to **not** apply any filtering based on application ID
 4. Press the trash can button to clear the existing log buffer
 5. Reproduce DS app crash
 6. Copy-paste the output in the window into a text file and upload it here

If you do not use Android Studio, we'll try to figure out another way to get the logs.

crowcasso, 2021-09-25:
[crash.txt](https://github.com/FIRST-Tech-Challenge/FtcRobotController/files/7229056/crash.txt)

crowcasso, 2021-09-25:
Attached text file above.

Windwoes, 2021-09-25:
Interesting, looks like some sort of permissions issue.

```
2021-09-24 22:42:41.952 7297-7297/? E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.qualcomm.ftcdriverstation, PID: 7297
    java.lang.RuntimeException: Unable to start activity ComponentInfo{com.qualcomm.ftcdriverstation/com.qualcomm.ftcdriverstation.FtcDriverStationActivity}: java.lang.SecurityException: com.qualcomm.ftcdriverstation was not granted  either of these permissions: android.permission.CHANGE_NETWORK_STATE, android.permission.WRITE_SETTINGS.
    
```

crowcasso, 2021-09-25:
On a clean install (side loading through adb install) it does ask for permission to access  "photos, media, and files" and then again for location. Only when we switch to the Control Hub Pairing does it crash (must be trying to gain permission and failing).  

Really old phone running Android 6.0.

cmacfarl, 2021-09-25:
Well that log is interesting.

The DS does not ask for android.permission.CHANGE_NETWORK_STATE or android.permission.WRITE_SETTINGS, because in none of the tech team's testing, nor in any of the documentation regarding target sdk migrations from Google, that the permission work has been based upon, has this been identified as necessary. 

But your log doesn't lie.   Your phone wants those and the DS hasn't asked for them.  This is going to take some thought and time to reconcile.  Can you please provide the carrier that this phone is tied to?

crowcasso, 2021-09-25:
The phone is running without a SIM/carrier (Global GSM Unlocked Phone). 

v6.2 of the DS works fine (with the obsolete warning showing).

cmacfarl, 2021-09-25:
Update: The DS does not _explicitly_ ask for CHANGE_NETWORK_STATE, but it is listed in the manifest, and it's classified as a normal permission, so that should be enough.  However, this stack overflow entry suggests that there's a bug in v6.0 https://stackoverflow.com/questions/32185628/connectivitymanager-requestnetwork-in-android-6-0, see the answer [here](https://stackoverflow.com/a/34175655/5051323).

Is it possible for you to update your phone to v6.0.1?

Note also that the code path for network connection changed between v6.2 and v7.0 to support the REV Driver Hub which is likely why you don't see this on v6.2.

crowcasso, 2021-09-25:
Will see what I can do ... but its telling me that there are no updates (https://android.stackexchange.com/questions/145703/android-not-updating-from-6-0-to-6-0-1-for-moto-g-3rd-gen)

crowcasso, 2021-09-25:
Automatic updating does not work. I looked for a US version of the XT1540 rom, but have not found one.

AlecHub, 2021-09-25:
> We are not able to reproduce this issue on Moto G 3rd Gen phones ...

I can reproduce the issue on a Moto G2 and on a Moto G3.   Both are running Android v6.0, no SIM card installed.  Simple "fix" is to make Moto G2 and G3 illegal for Freight Frenzy.

cmacfarl, 2021-09-25:
>  Simple "fix" is to make Moto G2 and G3 illegal for Freight Frenzy.

Which would make all moto g2/g3 phones that are on 6.0.1 illegal.   Not really an option.  Deciding which phones to drop support for is not something that the tech team takes lightly and we've never done it mid-season.  I think we have a root cause.  We'll work internally on a solution, but it will likely take a bit of time.  Thank you for helping to bring this to our attention.

crowcasso, 2021-09-25:
After some research it looks like Motorola only pushed 6.0.1 to some of the G3s. The ones we have (model XT1540)  were not upgraded beyond 6.0, so there is no easy upgrade path for this model of the G3 phones.

AlecHub, 2021-09-26:
> > Simple "fix" is to make Moto G2 and G3 illegal for Freight Frenzy.
> 
> ... Deciding which phones to drop support for is not something that the tech team takes lightly and we've never done it mid-season. I think we have a root cause. We'll work internally on a solution, but it will likely take a bit of time ...

If you have a Moto G2 or G3 that's stuck on Android 6.0, you are not going to wait around for the tech team to support your phone.  You have no choice but to switch to another phone.   So your Moto G2 or G3 is already effectively illegal to use with a Control Hub, mid-season.

NoahAndrews, 2021-09-26:
> If you have a Moto G2 or G3 that's stuck on Android 6.0, you are not going to wait around for the tech team to support your phone.  You have no choice but to switch to another phone.   So your Moto [...]

Kickoff was 8 days ago. Most teams are still brainstorming, and likely haven't done anything with their phones yet. Please don't assume that we're not going to have a solution in place in time for those phones to be useful. Teams can also use whatever non-legal phones they might have on hand for practice, until we get the fix out.

By the way, I developed said fix yesterday. It still needs to be tested and released of course, and I can't make any promises about exact timelines, but please don't make the unfounded assertion that these phones are "effectively illegal".

AlecHub, 2021-09-26:
> > > > Simple "fix" is to make Moto G2 and G3 illegal for Freight Frenzy.
> > > 
> > > 
> > > ... Deciding which phones to drop support for is not something that the tech team takes lightly and we've never done it mid-season. I think we have a root cause. We'll work internally on a solution, but it will likely take a bit of time ...
> > 
> > 
> > If you have a Moto G2 or G3 that's stuck on Android 6.0, you are not going to wait around for the tech team to support your phone.  You have no choice but to switch to another phone.   So your Moto G2 or G3 is already effectively illegal to use with a Control Hub, mid-season.
> 
> Kickoff was 8 days ago. Most teams are still brainstorming, and likely haven't done anything with their phones yet. Please don't assume that we're not going to have a solution in place in time for those phones to be useful. Teams can also use whatever non-legal phones they might have on hand for practice, until we get the fix out.
> 
> By the way, I developed said fix yesterday. It still needs to be tested and released of course, and I can't make any promises about exact timelines, but please don't make the unfounded assertion that these phones are "effectively illegal". We will get them fixed as soon as we can.

The bottom line is nobody cares if this doesn't get fixed.  **The only thing to be fixed is the optics.** So the fix is entirely self-serving on the part of the tech team and FIRST.

cmacfarl, 2021-09-28:
@crowcasso, and other Moto G series phone users whose phones will not upgrade past Android 6.0.  A new Driver Station APK has been attached to the v7.0 release page that fixes this issue.  Please see https://github.com/FIRST-Tech-Challenge/FtcRobotController/releases/tag/v7.0 and use the 7.0.1 version of the Driver Station.

This issue will remain open until the next official release so that other teams who run into this problem and come here looking for a solution may find this discussion.

Thank you for bringing this to our attention.

closed 2022-09-14 by cmacfarl