FIRST-Tech-Challenge/FtcRobotController #173: Driver Station Crashes with v1 PS4 Controller

ryleu, 2021-09-25:
When connecting a gen 1 PS4 controller to a phone running the driver station app (I currently only have access to a Moto G Play gen 1, though [`@NPE | OpenFTC Developer`](. "<@367074823605583872>") on Discord said it was a DS bug), the app crashes with this error:

```java
2021-09-25 15:44:15.551 26352-26352/? D/AndroidGamepadManager: onInputDeviceAdded() id=24 vid=0x54C pid=0x5C4
2021-09-25 15:44:15.554 1639-2738/? W/ContextImpl: Calling a method in the system process without a qualified user: android.app.ContextImpl.sendBroadcast:763 com.android.server.usb.UsbSettingsManager.deviceAttached:746 com.android.server.usb.UsbHostManager.endUsbDeviceAdded:246 com.android.server.usb.UsbHostManager.monitorUsbHostBus:-2 com.android.server.usb.UsbHostManager.-wrap0:-1 
2021-09-25 15:44:15.556 1639-2738/? I/ActivityManager: START u0 {act=android.hardware.usb.action.USB_DEVICE_ATTACHED flg=0x10000000 cmp=com.qualcomm.ftcdriverstation/.DriverStationUserspaceAlias (has extras)} from uid 1000 on display 0
2021-09-25 15:44:15.559 26352-26352/? V/DriverStation: Lifecycle onPause() : 8763187
2021-09-25 15:44:15.561 26352-26352/? D/DirectAccessGamepadMgr: ACTION_USB_DEVICE_ATTACHED_INTENT device id=1004 vid=0x54C pid=0x5C4 /dev/bus/usb/001/004
2021-09-25 15:44:15.561 26352-26352/? D/DirectAccessGamepadMgr: USB device id=1004 has been autodetected based on USB VID and PID as a Sony PS4 (Gen 1) gamepad; hooking with userspace driver
2021-09-25 15:44:15.562 26352-26352/? D/PS4GamepadImpl(id=1004): openAssumingPermission()
2021-09-25 15:44:15.562 26352-26352/? D/AndroidRuntime: Shutting down VM
2021-09-25 15:44:15.563 26352-26352/? E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.qualcomm.ftcdriverstation, PID: 26352
    java.lang.RuntimeException
        at org.firstinspires.directgamepadaccess.usb.ps4.PS4GamepadImpl.openAssumingPermission(PS4GamepadImpl.java:156)
        at org.firstinspires.directgamepadaccess.usb.DirectAccessGamepadManager.connectGamepad(DirectAccessGamepadManager.java:255)
        at org.firstinspires.directgamepadaccess.usb.DirectAccessGamepadManager.onNewIntent(DirectAccessGamepadManager.java:176)
        at org.firstinspires.directgamepadaccess.core.CompositeGamepadManager.onNewIntent(CompositeGamepadManager.java:77)
        at com.qualcomm.ftcdriverstation.DriverStationGamepadManager.handleNewIntent(DriverStationGamepadManager.java:486)
        at com.qualcomm.ftcdriverstation.FtcDriverStationActivity.onNewIntent(FtcDriverStationActivity.java:1085)
        at android.app.Instrumentation.callActivityOnNewIntent(Instrumentation.java:1235)
        at android.app.Instrumentation.callActivityOnNewIntent(Instrumentation.java:1247)
        at android.app.ActivityThread.deliverNewIntents(ActivityThread.java:2566)
        at android.app.ActivityThread.performNewIntents(ActivityThread.java:2578)
        at android.app.ActivityThread.handleNewIntent(ActivityThread.java:2587)
        at android.app.ActivityThread.-wrap12(ActivityThread.java)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1429)
        at android.os.Handler.dispatchMessage(Handler.java:102)
        at android.os.Looper.loop(Looper.java:148)
        at android.app.ActivityThread.main(ActivityThread.java:5459)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:728)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:618)
2021-09-25 15:44:15.566 1639-3369/? W/ActivityManager:   Force finishing activity com.qualcomm.ftcdriverstation/.FtcDriverStationActivity
```

This only happens when advanced gamepad features are enabled. If the features are disabled, the controller works perfectly.

Android version: 6.0.1
App version: 7.0.0
Hardware: Moto G Play
Controller model number: CUH-ZCT1U

I'm willing to help test software if the developers don't want to purchase a gen 1 PS4 controller.

Windwoes, 2021-09-25:
Thank you for the report, this is definitely a bug in the Driver Station. We'll take a look.

Windwoes, 2021-11-08:
This will be fixed in the next release of the Driver Station / SDK.

ryleu, 2021-11-08:
> This will be fixed in the next release of the Driver Station / SDK.

Thank you, I appreciate it!

Windwoes, 2022-01-17:
Fixed in 7.1

closed 2022-01-17 by Windwoes