FIRST-Tech-Challenge/FtcRobotController #176: Tablet Needed for the Visually Impaired

AlecHub, 2021-09-27:
A lot of the older team members (and some of the younger team members) have great difficulty reading telemetry on a phone or REV Driver Hub.  We request that several models of Android tablets be tested and approved for use as a Driver Station.

AlecHub, 2021-09-29:
## 14-Inch Android 10 Tablet for $220:
- **Two (2) USB Type-A ports**
- 5GHz/2.4GHz WiFi
- 1.3 GHz Quad-Core Processor / 3GB RAM / 64GB Storage
- Attachable Keyboard (OnBot Java Programming from the Driver Station)
- Dedicated DC power/charging port
- **In Stock**

### [Venturer VCT9T48Q34RBM](https://www.amazon.com/dp/B08PHL7NLF/ref=cm_sw_em_r_mt_dp_AF17BG0JV4MYMGA68CVA?_encoding=UTF8&psc=1)

texasdiaz, 2021-09-30:
Thank you for your suggestion, we'll take this under consideration during our next hardware evaluation period.

In the meantime, we recommend trying to adjust the "Font Size" and "Display Size" display settings to increase the size of displayed elements on your device. The real difference between these two settings is that "Font Size" only affects text, while the "Display Size" affects both text and UI elements. For the largest possible text both of these values would be maximized, but you may want to find the right balance between size and and display space for your team. We realize this isn't a perfect solution, but perhaps it could be helpful.

Here are the steps to change those settings:
1. Open the Settings app
2. Navigate to Display
3. Expand the Advanced section (not necessary on all devices)
4. Tap the Font size and/or Display size settings, and select your desired value.

AlecHub, 2021-09-30:
> Thank you for your suggestion, we'll take this under consideration during our next hardware evaluation period...

Great, thanks!  

For now we can use a Tablet for development, scrimmages, pits, outreaches, etc., which is the vast majority of the time that we use a Driver Station.  We will switch to a phone when running matches at a traditional event, which is a small fraction of the time.

So this request is more of a software request, namely to add tablet layouts to the Driver Station App.

AlecHub, 2021-10-03:
## Driver Station 7.0 Running on a 14-Inch Tablet
### Paired with Robot Controller running on a phone:
[image]
**Left side of Landscape layout is mostly wasted space when rendered in 1920 x 1080.** 
### Blocks editing within Driver Station:
[image]
### OnBotJava editing within Driver Station:
[image]