FIRST-Tech-Challenge/FtcRobotController #181: Javadoc link is broken on github page

pmichaud, 2021-09-30:
When I click on the link "FTC Javadoc Documentation" on the FtcRobotController home github page (in the README.md under "Javadoc Reference Material"), the resulting page gives an error saying "/org.firstinspires.ftc is not valid".

Pm

pmichaud, 2021-09-30:
I'll note that this is apparently the same issue as reported in #177 -- I didn't see it reported there when searching issues because it's listed as "closed".

Feel free to close this ticket as a duplicate, but since the issue still exists it might be worth keeping one of these two open.

Also, does fixing a README.md file on GitHub really need to wait for a 7.1 release...?  That seems odd -- seems like a simple commit to README.md would fix it at least as far as GitHub users are concerned.  (I can also imagine there's a workflow where this isn't possible.)

Pm

cmacfarl, 2021-09-30:
Yeah, you beat me to it.  I'll leave this open until v7.1 is out.

Pending a fix in v7.1, in the meantime please see https://javadoc.io/doc/org.firstinspires.ftc

> Also, does fixing a README.md file on GitHub really need to wait for a 7.1 release...? That seems odd -- seems like a simple commit to README.md would fix it at least as far as GitHub users are concerned. (I can also imagine there's a workflow where this isn't possible.)

Indeed.  There are two issues, 1) All the files in this repository originate from somewhere else, and so a change here easily gets lost in the originating location and humans, being mistake prone, are likely to end up with out of sync files if said humans started modifying them on the public repository, and 2) the goal is to keep the commit history on the public repo as clean as possible.  Ideally there are only commits for releases, although I realize that goal wasn't always met in the past.  

Issue (1) is the basis for the often made statement that the public repository's release model does not allow for public pull requests against this repository.

AlecHub, 2021-10-03:
Also, [CONTRIBUTING.md](../blob/master/.github/CONTRIBUTING.md) has outdated links and URLs, like **ftcforum.usfirst.org**, **ftctechnh/ftc_app**, etc.  

Use relative links when linking within GitHub.

AlecHub, 2021-10-03:
> Indeed. There are two issues, 1) All the files in this repository originate from somewhere else ...

So in essence the repo contains only "generated" "code".  Perhaps all this should be explained in [README.md](../blob/master/README.md) and/or in [CONTRIBUTING.md](../blob/master/.github/CONTRIBUTING.md).

closed 2022-11-22 by CalKestis