FIRST-Tech-Challenge/FtcRobotController #217: API to Query Color Scheme Selection of Driver Station

AlecHub, 2021-10-30:
We would like our OpModes to be able to query the Color Scheme Selection of the Driver Station.  Of particular interest to our OpModes are the Blue and Red color scheme selections: 

[image]

The Dirver Station color scheme selection could be sent to the Robot Controller as part of the Init Message (the message that gets sent to the Robot Controller when the driver taps INIT).

Also, it would be nice to be able to change the Driver Station color scheme without having to restart the App.

Windwoes, 2021-11-01:
Out of curiosity what is the desired use case?

alan412, 2021-11-01:
I am not the OP, but my guess is have the students set the color scheme of the driver station based off of their alliance and then the robot can select the correct autonomous to run based off of which alliance it is on.

AlecHub, 2021-11-01:
> I am not the OP, but my guess is have the students set the color scheme of the driver station based off of their alliance and then the robot can select the correct autonomous to run based off of which alliance it is on.

That is one possible use case.  I wanted to be vague so that I don't steer teams away from use cases that may be higher priority for them (although "Red" and "Blue" are hardly vague ;)

**My personal preference is the opposite approach:**  allow the OpMode to dynamically change the Driver Station Color Scheme to any 32-bit color as a form of Telemetry or status indication.  Or better yet, a bi-directional approach (`getDSColor()`, `setDSColor()`), analogous with many of the other APIs.

alan412, 2021-11-01:
For what it's worth, I **love** the idea of being able to set from the robot.  We typically allow the drive coach to set the alliance color during init and that lights up LEDs on the robot showing which alliance.   Also showing on the screen would be pretty neat.

AlecHub, 2021-11-02:
> Out of curiosity what is the desired use case?

I understand and appreciate the question.  My thinking is we should let teams get creative with the feature.

cmacfarl, 2021-11-02:
Themes are immutable once applied, which is why the applications require a restart after changing the color theme.  I don't think this has changed within the Android ecosystem since the feature was implemented.